BCN3D/BCN3DSigma-Firmware #115: SD card not detected

lampi87, 2017-02-23:
Hello,

I was able to print some pieces but now printer does not recognize inserted SD card. Even with inserted SD card the screen "Insert SD card" is shown.
I already tried it with different file systems: FAT, FAT16, FAT32, exFAT but it does not solve the problem.
Any ideas?

Thx,
Alexander

AlejandroGarcia92, 2017-02-27:
Maybe the SD card module is not working properly

Zorgonaute84, 2017-03-09:
I have same issue since the upgrade 1.2.2

AlejandroGarcia92, 2017-03-10:
@Zorgonaute84 Can you share us a Gcode to analyse the headers or other issues?

Zorgonaute84, 2017-03-13:
Attached a Gcode generate from Cura 2 :
[Rook.gcode.zip](https://github.com/BCN3D/BCN3DSigma-Firmware/files/838006/Rook.gcode.zip)

AlejandroGarcia92, 2017-03-13:
@Zorgonaute84 Cura 2 gcode headers seems not to be compatible with the firmware 1.2.2 as I tested it. There are two solutions: to use our Cura version using our headers or to try our release candidate firmware version [v01-1.2.3RC](https://github.com/BCN3D/BCN3DSigma-Firmware/tree/v01-1.2.3RC) which will be released at the end of the month. Another thing to take into account is that RC version only obtains the printing info(printing time and material) version from our Cura or Simplify3D. So ,in your case, if you try to use the RC version, it will appear 1min of printing time.

AlejandroGarcia92, 2017-03-13:
[Release notes](https://github.com/BCN3D/BCN3DSigma-Firmware/wiki/1.2.3)

closed 2017-04-05 by AlejandroGarcia92