BCN3D/BCN3DSigma-Firmware #118: Bed leveling issue since Stepper driver upgrade kit

Zorgonaute84, 2017-03-09:
Hello,

I've just upgraded my sigma with the fan and stepper driver kit. My new printer is very quiet, but it is non operationnal cause of a bug with my bed leveling.

Before the upgrade, the homing processus looks something like : 
- Move the left extruder to position the end course sensor
- Level up the bed until it reach the sensor
- Level down the bed to 5 or 10mm

after the homing processus, when you want to print something the bed is level up to same amount that it level down during the homing processus (5 or 10mm).

Now my homing processus stop when the bed reach the sensor. So the bed doesn't perform the level down describe just before. The problem is, when I want to print something, the bed is already at the 0 position, but the bed level up to few millimeters and my nozzle hit the glass !!

I occured a similar issue during the bed calibration. My bed stay at the position 0 during all the process. I attached a video of the bed calibration process.

Thanks for your help !!
[Video of Calibration process.zip](https://github.com/BCN3D/BCN3DSigma-Firmware/files/830253/Video.of.Calibration.process.zip)

dds4pc, 2017-03-10:
Hello,

I have the same problem after stepper and coolfan upgrade.
Any help yet?

Zorgonaute84, 2017-03-10:
In my case, the support send me a new Driver. Wait and See.
For the moment, I mount my old drive for the Z axis.

You just upgrade the drivers ? After replacing the Z driver by an old one, could you tests some prints ? I have some issue. The printer is very quiet, but I have extrusion issue (a lot of missing step) ans I had one drifting of the X axis (left head). I supposed it was a missing step too.

dds4pc, 2017-03-10:
Contacted support and it looks like they are going to send me another one.
Replaced old Z stepper and all ok.

dds4pc, 2017-03-11:
Looks like it is not ok. Underextrusion on first layers and X axis skip.

[image]

Zorgonaute84, 2017-03-11:
Exactly like me !! The underextrusion and the X axis skip is due to missing step. During extrusion do you hear some "Poc". That's the sound of à missing step.

kitusmark, 2017-03-13:
Hi @Zorgonaute84 
Can you post a photo of the new Z driver which does not work?
What configuration files are you guys using? ProGen?
Check the purge container and watch out for possible collisions with the rubber pad.

Zorgonaute84, 2017-03-13:
Hello Marc,

Please find attached 2 pictures of the Z driver.
Mainly I'm using Cura for Dual extrusion and Simplify3D for dual extrusion with my own setup.
My purge container is always empty. I'm using a "big" retraction and prime tower in my changing tool process. And I havn't see any collision. 

Sorry but, except replaced my old drivers by the new one I didn't make any change in my setup and my print process. Before the new driver I hadn't shifting on any axis and no underextrusion trouble caused by missing step.

dds4pc, 2017-03-13:
My stepper has the same production date and rev.
I have removed all my new steppers till problems are solved.

dds4pc, 2017-03-15:
Any news about this?

Zorgonaute84, 2017-03-16:
No... My new drivers will be delivered this afternoon. I'll keep you inform.

dds4pc, 2017-03-16:
Do they replace only one or all 4?

Zorgonaute84, 2017-03-16:
I received 3 as planned. One for the Z driver and 2 for the extruders (One was out if service and the second with the missing step).

They are not mounted yet, I'll try this week-end.

dds4pc, 2017-03-17:
How will the drifting on the x axis be fixed then?

Zorgonaute84, 2017-03-17:
For the moment I did nothing for the X axis drifting. One problem at a time. I think it could be fixed by adjusting the potentiometer but I'm waiting advice from BCN3D before try something.

kikolobo, 2017-03-18:
I do hear the POC after upgrading to the new drivers. Will contact support. My guess is the drivers are not working as expected!

Zorgonaute84, 2017-03-20:
Finally I change my stepper driver by the new one I just received from BCN3D.

1/ With the new driver the bed leveling issue disappear !!! (I also updated to 1.2.3 RC firmware)
2/ The bed still noisy when it move fase (Z homing for example) but very quiet during changing layer.
3/ The missing step on the right extruder look disapeard too ! Good news !!! I'm printing at this moment, so wait and see...
4/ The right extruder doesn't work properly ! I have the exact same issue on the right extruder that I have when I plugged it the first Z driver. Meaning that the extruder only working in one way. It can Retract only. Even if I want an extraction, the driver retract... Very strange issue...

closed 2017-09-19 by AlejandroGarcia92