BCN3D/BCN3DSigma-Firmware #120: Print Status not displayed when started from PC

Kevin-Stager, 2017-03-10:
I am using the latest released firmware and Simplify3D.  If I start a print from the SD card with Simplify3D the normal printing screen shows up but all the values are left blank. I.E. Temperatures, print percentage, speed. 

However if you pause the print from the printer's touch screen and then resume it the correct values are shown or the rest of the print. Except that the print time, hour value is in the thousands and never changes. The print percentage is correct and updates.

AlejandroGarcia92, 2017-03-10:
Can you share us a Gcode to analyse the headers or other issues?

Kevin-Stager, 2017-03-10:
Attached a full gcode file. 
I do not have a M117 Command like cura normally starts with so maybe that is the cause?
But I think that is only for the message. 
I will test it when I get a chance.

The only change I made to the simplify3D default for the sigma is a line to set the max acceleration. 
Should probably put that in as a different bug but the max acceleration, M204, was set to "nan" several times. I was not sending any intentional M codes when it changed to nan. Might have been caused by printing over usb from simplify3d or from doing the automatic hotend pid tune.

This meant the machine used no acceleration limit for G0 moves, including bed leveling which is unusable. Fortunately it is as simple as resetting the acceleration limit to fix. 

[Z Conduit Nut Trap.txt](https://github.com/BCN3D/BCN3DSigma-Firmware/files/834741/Z.Conduit.Nut.Trap.txt)

AlejandroGarcia92, 2017-03-13:
I have run successfully this Gcode via SD and the info is shown. M117 doesn't affects for sure and "nan" means not a number.

Kevin-Stager, 2017-03-13:
Did you start the print from Simplify3D? with the "print from sd card" button? That is when the info is missing. If I start the code from the printer it shows the info. 

Yes I assumed the acceleration max was corrupted or changed to a invalid value somehow which is why it was not a number and ignored. Just don't know how it is getting changed.

AlejandroGarcia92, 2017-03-14:
@Kevin-Stager I misunderstood it. The problem exist because the firmware has not implemented to support this feature. I will develop this feature for the next update. First in the release candidate version.

Kevin-Stager, 2017-03-14:
Ok. The part I found odd was that pausing and resuming the print from the printer screen fixes it.

AlejandroGarcia92, 2017-03-20:
Up [here](https://github.com/BCN3D/BCN3DSigma-Firmware/tree/v01-1.2.3RC).

closed 2017-04-05 by AlejandroGarcia92