BCN3D/BCN3DSigma-Firmware #121: Bed leveling freezes when PC is connected

Kevin-Stager, 2017-03-10:
If the printer is connected to simplify3D and I run the bed leveling program from the sigma touch screen it does the first touchdown then the LCD freezes and never continues. 

If it is not connected bed leveling seems to work fine.

This may only be when the printer is executing a heat command from the PC.
I usually start the bed heating than level it once it is close to temperature.

AlejandroGarcia92, 2017-03-10:
The simplify3d verbose communication collapses the marlin buffer and it doesn't let to execute the bedleveling program. So I suggest to disable the verbose.

Kevin-Stager, 2017-03-10:
Ok I tried that and the problem still occured. I will try changing some of the other simplify3D communication settings.  Right now I have no flow control and command buffering is turned on. 
 
The work around of just unplugging before auto leveling is easy enough for now.

kitusmark, 2017-03-13:
@Kevin-Stager Maybe this helps.
https://github.com/BCN3D/BCN3D-Cura-Windows/wiki/Connect-the-BCN3D-Sigma-to-Simplify-3D

AlejandroGarcia92, 2017-04-05:
Dear @Kevin-Stager, in the new release should be fixed.

closed 2017-09-19 by AlejandroGarcia92