BCN3D/BCN3DSigma-Firmware #123: Short pauses during print

m0w1337, 2017-04-05:
Since the Update to Version 1.2.2 I have the problem, that the printer stops randomly for about 1 second before printing continues. The print is OK, but it results in blobs that are not very nice looking. I remember this was already an issue in earlier versions but was fixed until at least version 1.2.0 with which i didn't observe this problem.

AlejandroGarcia92, 2017-04-06:
@m0w1337 the issue that you are describing is similar to the version 1.1.5. We fixed in the 1.1.6 and this problem has not appeared anymore. So we have not change any in the firmware that could affect in this way. One thing more, the stops were about 0.1 seconds aprox.

m0w1337, 2017-04-06:
Hello,
I definitely see at least 0.5seconds stops (but I'll try to measure next time) and mainly when printing round objects I'll try different SD cards and will have an eye on that. Thank you very much for the quick reply.

kikolobo, 2017-04-16:
I can confirm I see this behavior happen in 1.2.2.. Haven't upgraded to 1.2.3

rtrolle, 2017-06-06:
If you are printing from USB, GCODE is being sent constantly over serial to the machine.  What I noticed, if you are using your computer for other work (Meshing, CAD, etc), you easily reach 100% CPU, and I have gotten the nozzle to do what you have described.  Running 1.2.2.  

I haven't seen this behaviour in printing from SD.

m0w1337, 2017-06-07:
Well I have never printed via USB because i know about the small buffers. I always have this issue when printing via SD Card, the type of card does not matter I tried several speeds and manufacturers.

WanaGo, 2017-06-22:
I am on 1.2.4 and this happens to me also. Its printing fine, and then maybe in a 4 hour print, it will pause for a second maybe 2-3 times over the course of the print. Noticeable as I have this in the same room as me at the moment, and its very noticeable as everything goes quiet for a moment. I havent noticed an problems due to it however, yet. SD card loading, not over USB.