BCN3D/BCN3DSigma-Firmware #128: 1.2.4 SD files for replicator/mirror mode

lemmans, 2017-05-23:
Hi,

Can you put the SD files to activate replicator and mirror mode present in the last firmware 1.2.4?

If right now is only possible by GCode command, can you post a GCode example using replication or mirror mode?

Thanks!

AlejandroGarcia92, 2017-05-24:
Replicator/mirror mode, both, are disabled in the official release.

lemmans, 2017-05-24:
So, if I install 1.25 beta is this function available?
How can I activate, use it? Is there any specific SD screen files?
Any Example.

Thnaks again for your efforts

AlejandroGarcia92, 2017-05-26:
@lemmans you are able to install 1.2.5 beta where those modes are enabled. It is not LCD screen files change required. In the upcoming week, I'm going to make a post in the wiki explaining how to activate properly because there are some issue that can happen and you, as a user, must take into account.

lemmans, 2017-05-29:
Thanks! wi will wait for this wiki.

m0w1337, 2017-08-02:
Hi,

this possibility would be very helpful for me as well. Is there any place to find the wiki entry, or can you maybe tell me the beta version that had these functions enabled?
Thanks for keeping up the good work!

closed 2018-10-02 by AlejandroGarcia92