BCN3D/BCN3DSigma-Firmware #129: 1.2.4  - stepper motors getting extremely hot?

tsp666, 2017-05-26:
Hi,

I just noticed that the stepper motors are getting extremely hot, so they cannot be touched since the update, is that possible ?
I tried to print with different objects, but always the same.
It's a upgraded R17 .

Many thanks,
Torsten

kitusmark, 2017-05-29:
Hi @tsp666 

The change in Firmware can not possibly modify the power the driver outputs to the motors. Each driver has a potentiometer that sets the max current per winding, and it is fixed if you did not touch it. 

FYI the motors do run hot. It is safe to run them to around 60-70 ºC .

tsp666, 2017-05-29:
Hi,

I never noticed they are getting that hot before. Is it normal that even motors NOT USED are heating up to that level? (e.g. from extruder not used)

kitusmark, 2017-05-29:
They are not moving but they are used as they need to be blocked to maintain the position. Do you have any way to measure the temperatures ?

tsp666, 2017-05-29:
Temp is 62 Degrees Celsius, so It never came to my attention even after 1400 hours of printing with the sigma...

closed 2017-05-29 by tsp666

kitusmark, 2017-05-29:
In my opnion that temperature is completely normal.