BCN3D/BCN3DSigma-Firmware #130: Can't get Z calibration right

ctsvette, 2017-05-27:
Firmware V01-1.2.3
Machine BCN3D Sigma 2015
Bed Calibration or full calibration do not give proper Z calibration.
Printer is more than 1 cm away from the bed.
Impossible to print anything as filament won't stick to the bed because of excessive distance.
Redone the full calibration 5 times. Went through the setup assistant twice.
Nothing works. Manual calibration can't adjust such a distance. 
Please help

RhepTheSpring, 2017-05-27:
Z Callibration doesent seem to be helping re calibrated works one day next day crap. re calibrated start print can even fit paper under nozzle. while printing.

AlejandroGarcia92, 2017-05-29:
@ctsvette you are able to fill manually by serial connection the calibration values [M530](https://github.com/BCN3D/BCN3DSigma-Firmware/wiki/Tips#2-m-code-m530-is-useful-to-change-the-calibration-values). Anyway, can you share us what are your current calibration values by using [M503](https://github.com/BCN3D/BCN3DSigma-Firmware/wiki/Tips#3-m-code-m503-show-us-the-configuration-values). Take into account that the z-prove offset represents the distance between endstop and nozzle and Z represents the height distance between nozzles.

AlejandroGarcia92, 2018-01-30:
In v01-1.2.7 you will be able to go up to 5mm. Until now, it was bounded restricted in 1.5mm approximately.

closed 2018-01-30 by AlejandroGarcia92