BCN3D/BCN3DSigma-Firmware #131: 1.2.4 head crash when doing full calibration

droomurray, 2017-05-29:
when doing a full calibration and doing Y calibration the software crashes the heads into the front of the printer when extruding the 4th line using the 2nd extruder (on the right)

kikolobo, 2017-06-04:
I have the same issue with 1.2.3

Is this persistent in your case or is it intermittent? In my case 2 consecutive calibrations and it keeps doing it.

droomurray, 2017-06-04:
Mine was fine on 1.2.2 and 1.2.3 but now on 1.2.4 it does it every time.

Its when it does the y alignment routine of the right extruder, sometimes will draw the first 3 lines sometimes only 1 line.

It will just after drawing a line move the carriage full speed to the front and GGRRRRZZZ noise !

AlejandroGarcia92, 2017-06-06:
We are aware of that issue but this is not an issue that can be fixed by an update because as we investigated this is not a firmware problem. Contact Support.

droomurray, 2017-06-06:
oh No, I was hoping it would be a simple firmware fix.

haqbany, 2017-06-13:
no, its firmware fix, I just downgraded to 1.2.3 any everything works fine after

AlejandroGarcia92, 2017-07-27:
Hi all, were you using our compiled firmware version or were you trying to compile the firmware by your own?. Because when I recently updated my arduino IDE to 1.8.3 this issue happens to me. So, I still using arduino 1.6.8 IDE version.

thesec, 2017-10-06:
Just happened to me after updating to my own 1.2.6 using arduino IDE 1.8.5.
Is this a cause of the IDE?! So what is the root cause?
Are you still using 1.6.8 @AlejandroGarcia92 ?

AlejandroGarcia92, 2017-10-09:
@thesec I compiled 1.2.6 with IDE 1.6.8 and the board is atmega2560. Since a few days, I'm testing IDE 1.8.4 and I feel that works fine. I don't know the root cause but by changing Arduino IDE version works for me.

AlejandroGarcia92, 2017-10-09:
#162

exitgames, 2017-10-09:
Thank you @AlejandroGarcia92. Getting fine builds with arduino 1.8.5 now.
My "head crash" was not related to the compiler but old calibration offsets still stored from prior belts' maintenance ... once removed the crash was gone. So -1 on this issue ;)

thesec, 2017-10-09:
Thank you @AlejandroGarcia92. Getting fine builds with arduino 1.8.5 now.
My "head crash" was not related to the compiler but old calibration offsets still stored from prior belts' maintenance ... once removed the crash was gone. So -1 on this issue ;)

AlejandroGarcia92, 2017-10-30:
Line 4174 Marlin_main.cpp:
```
		if (i != 9){
			#if BCN3D_PRINTER_SETUP == BCN3D_SIGMA_PRINTER_DEFAULT
			current_position[X_AXIS] = mm_right_lines_y_r;
			#else
			current_position[X_AXIS] = mm_right_lines_y_r + X_OFFSET_CALIB_PROCEDURES;
			#endif
			current_position[Y_AXIS]=y_first_line-((i+1)*mm_each_extrusion)-mm_second_extruder[i+1];
			plan_buffer_line(current_position[X_AXIS],current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS], 200, active_extruder);
			st_synchronize;  <---------- Bug here
			if(gif_processing_state == PROCESSING_ERROR)return;
		}
```
This bug is now fixed for the next releases. https://github.com/BCN3D/BCN3DSigma-Firmware/releases/tag/01-1.2.7RC_M26.10

closed 2017-10-30 by AlejandroGarcia92