BCN3D/BCN3DSigma-Firmware #133: Feature idea: X:Y:Z calibration

tsp666, 2017-06-07:
Hi,

when printing a cube with x:y:z of 50mm, I noticed that the result in the 3 axes are not equal. Results can look like the following ones:

x=49.75 mm
y=49.36 mm
z=50.30 mm

Would it be possible to add a scaling factor to the firmware to equalize this, based on a printed object?
Attached the sample cube.

Many thanks
Torsten
[cube50mm.stl.zip](https://github.com/BCN3D/BCN3DSigma-Firmware/files/1057021/cube50mm.stl.zip)

WanaGo, 2017-06-17:
Please, I second this. Trying to print dimensionally accurate parts and they simply dont fit. Out by 0.4 to 0.8mm or so it seems. I means we cant prototype accurately as x and y are too small but z is too large.

4truemodels, 2017-08-17:
Mmmmm, very good idea, maybe they can develop an automatic fine callibration order that prints a cube test for the user to mesure it and input the results in X, Y and Z so the Sigma will save them and change the propper values in the firmware to solve that (the number of steps per each motor).