BCN3D/BCN3DSigma-Firmware #134: [Bug] Z axis calibration wanders

JoeJEngineer, 2017-06-16:
Since the latest firmware update I have had to adjust the Z axis to be higher using the manual calibration after every print.  I'm at .475 now after about 8-10 prints.  I'm on the latest firmware and I upgraded the stepper drivers recently but it's done this before and after driver upgrade.  Doing a full calibration resets it to around zero but it immediately starts wandering after that.

axiiiofevil, 2017-08-14:
I concur with this. Mine is wandering too...

svengl, 2018-02-13:
I cannot really reproduce it, but I also have the feeling that I calibrate and it works fine. And then after some time the z-axis wanders and I get bad first layers and don't understand why because it had worked just in the print before. Then I either add manual correction or have to recalibrate. 
Is there a way to see in gcode what the calibration result is? That might help to see if there is some shift. Or maybe it can even be restored?