BCN3D/BCN3DSigma-Firmware #135: Full Calibration but skip first test, hotend doesn't pre-heat

WanaGo, 2017-06-18:
Hello,

When doing a full calibration, I skipped the first test which was for the Z-Axis, as I had just done that a few minutes before and it was all OK. It then started to do the X-Axis calibration, and immediately started to draw the lines, however it didn't first heat the hot ends. So the system was trying to force the filament through the hot end which was only partially warm. This obviously resulted in some grinding and nothing on the glass. Doing a 'Redo', and it went through the process correctly.
Seems there is a step missing in the program that it didn't check prior to going into this 2nd calibration step.

I have also noticed when it drew the lines for the X calibration, that it didn't lift the nozel high enough as it was travelling home again, and it knocked each of the lines on its way back, moving them all.
I did not have any glue/spray on the table for the test, which would have contributed, however the nozel should not have been so low that it smashed into each of the single lines and moved them. surely...

Running latest Firmware and Display application.

AlejandroGarcia92, 2017-06-19:
I will check what happens. I give you an answer ASAP.

AlejandroGarcia92, 2017-06-27:
I've just make some changes when the hotends and bed are heating up to ensure that it reach the properly temps before printing. Dev branch here [01-1.2.5RC](https://github.com/BCN3D/BCN3DSigma-Firmware/tree/v01-1.2.5RC)

AlejandroGarcia92, 2017-09-19:
I found recently where was the problem on it. I'm doing some testing right now. This issue should be fixed on v01-1.2.6 version.

AlejandroGarcia92, 2017-10-30:
Fixed

closed 2017-10-30 by AlejandroGarcia92