BCN3D/BCN3DSigma-Firmware #136: Display stuck on 'Processing' at the end of a Calibration session

WanaGo, 2017-06-22:
Hello,
I have had this happen to me 3 times now, where you go to do a full calibration and it finishes, I can see the log in the USB that everything was completed, but the screen is stuck on Processing. The processor seems to still communicate, however the display does not return Home or wherever it should go.
Processing animation is not moving, its a static image at this point.

[image]

AlejandroGarcia92, 2017-06-26:
I will take into account for next firmware versions. The current dev branch is the [01-1.2.5RC](https://github.com/BCN3D/BCN3DSigma-Firmware/tree/v01-1.2.5RC)

ZAMHelmets, 2017-08-10:
I'm seeing this in a number of situations - the most common is loading/unloading filament, occasionally on calibration and once while trying to pause and restart a print (this was the most painful as the 10 hour job had to be restarted). In each case, the printer has to be restarted. Thank you for your consideration.

WanaGo, 2017-08-17:
Happened again. Full calibration, left extruder cleaned, paper pulled, drew its box and lines, all OK. Right extruders turn, cleaned the nozzle, table comes up ready to do paper pull, stops, and screen is stuck on Processing.
Please fix this!

v01-1.2.5 release version.

WanaGo, 2017-08-17:
Power cycled, started calibration again, it did it again! this time, left extruder, rising the bed up to do the paper pull after cleaning the nozzle.

WanaGo, 2017-08-31:
I get this all the time in calibration still. Sometimes though, its for a few seconds only. It can me moving the Z axis towards the hot end, and just pause for a few seconds, and then continue, or after moving the X towards the front of the printer, and then pause for a few seconds, and then resume. Something in the Marlin Firmware is blocking and its getting stuck.

norbertut, 2017-09-21:
I have exactly the same issue, LCD stuck on "processing" when I run a full calibration. It just finishes the bed calibration and then gets stuck.
I can run a bed calibration alone and it works well.
Release 1.2.5, just updated firmware from 1.2.2 and started having this issue.

RhepTheSpring, 2017-09-21:
I think this a well documented bug and needs fixing

AlejandroGarcia92, 2017-09-22:
This bug is planned to be fixed in the next release, hopefully during October's first week. Here https://github.com/BCN3D/BCN3DSigma-Firmware/tree/v01-1.2.6RC there is a release candidate version which is the firmware that we are testing.

AlejandroGarcia92, 2017-10-06:
Right now, I'm testing a new library for the LCD screen and it seems to run better than the old library. Making some debug communication(using new lib) LCD-mainboard, I realized, when screen is getting stuck, that marlin got a timeout on the communication between LCD-mainboard and came back online again.

```
Serial Debugging:
*** Genie WriteObject *** --->20... ACK !
*** Genie WriteObject *** --->20... ACK !
*** Genie WriteObject *** --->20... ACK !
*** Genie WriteObject *** --->20... ACK !
*** Genie WriteObject *** --->20... ACK !
*** Genie WriteObject *** --->20... ACK !
*** Genie WriteObject *** --->20... ACK !
*** Genie WriteObject *** --->20... Timeout!
*** Genie Disconnected from DoEvents! ***
*** Genie Connected / Sync! ***
*** Genie WriteObject *** --->20... ACK !
*** Genie WriteObject *** --->20... ACK !
*** Genie WriteObject *** --->20... ACK !
*** Genie WriteObject *** --->20... ACK !
*** Genie WriteObject *** --->20... ACK !
*** Genie WriteObject *** --->20... ACK !
*** Genie WriteObject *** --->20... ACK !
```
I guess that old lib when gets a timeout doesn't recover the connection with the LCD screen again.

I don't know, right now, why timeout is produced more frequently in some printer than others.

I'm still investigating.

thesec, 2017-10-07:
+1 on 1.2.6 on a R17

WanaGo, 2017-10-09:
#162 - Fingers crossed

AlejandroGarcia92, 2017-10-30:
https://github.com/BCN3D/BCN3DSigma-Firmware/releases/tag/01-1.2.7RC_M26.10

closed 2017-10-30 by AlejandroGarcia92

ZAMHelmets, 2017-11-13:
I took the release candidate from [here](https://github.com/BCN3D/BCN3DSigma-Firmware/tree/v01-1.2.7RC) as the link above didn't work for me. The LCD still freezes on me during the full calibration process. Steps to reproduce:

1. Click on Full Calibration from the LCD
2. Run z-axis calibration on both extruders (though I did it once skipping this step)
3. Run x-axis calibration
4. Redo x-axis calibration

The LCD freezes and says "Printing Test" after trying to redo the x-axis calibration the 2nd or 3rd time. This happened two times in a row and haven't had time to try a third. The first time I waited overnight and the LCD screen was still frozen the next day. Without the printer calibrated, I'm unable to run any additional prints to know if this is a more general issue. HTH.