BCN3D/BCN3DSigma-Firmware #137: Smart Purge ?

Zorgonaute84, 2017-06-30:
Is anybody could explain how work the smartpurge ?

Because I' using the script generate from ProGen, I also have my Sigma updated, but the smart purge during tool changing process look not purge anything... 
Thanks for your explanation.

guillemavila, 2017-07-03:
SmartPurge function allows you tu purge just the right amount of filament according to hotend idle time. It works in combination with the firmware, and ProGen as it needs to know material, hotend and idle time.

It's still in beta state and it comes commented (disabled) by default in ProGen toolchange scripts. If you want to test it just comment the default purge line and uncomment the other one.

If you already did it and it looks like not purging at all, maybe it's because idle time is too low.
Thanks for your feedback.

mrcg, 2017-07-16:
Such features need to be documented much better. Nobody knows about this and it's also difficult to use if not described. I would recommend to list all features on the progen webpage (seperated by beta/stable) from where the user can get all the details about how to use it.
Besid this it would be good to have option in the ProGen to enable/disable different features to be activated/deactivated in the generated profile.

guillemavila, 2017-07-19:
Yes, as it's still in beta stage and not enabled by default we just shared it is available in the [Firmware 1.2.3 release notes](https://github.com/BCN3D/BCN3DSigma-Firmware/releases/tag/01-1.2.3). I added what the parameters actually do internally.
Better documentation will be shared with new ProGen which enables the feature by default.

The main idea behind our ProGen tool is to deliver base profiles that just work, so people don't have to worry about any specific parameter more than machine Hotends and filaments used.
At the same time we didn't want to forget about our most advanced users so we keep them as fully editable profiles that you can tweak as much as you want in each Slicing software we support.

Thanks for the feedback!

closed 2017-10-30 by AlejandroGarcia92