BCN3D/BCN3DSigma-Firmware #139: Update 1.2.5 broke Simplify3D USB print capability

ShodanOne, 2017-07-24:
Hello,

I am unable to print via USB using Simplify3D since my update to 1.2.5. Simplify3D just can't connect to the printer ...

AlejandroGarcia92, 2017-07-24:
Hi @ShodanOne, I've realized that it's caused by the extra delay that I added whe the machine starts on this release. So, I've increased the waiting time when turn on the printer allowing more time to load the data screen files from SD card to Screen memory, it could help if the SD card is slower.

It seems that Simplify3D waits for a kind ACK at the start of the communication and the new delay at the start causes a timeout. I will take into account for the next release. By the way, you can disable this "ack waiting" on _Tools/Firmware Configuration/Communication/Communication Settings_ and   disable _Wait for startup command_.

It works for me, tell me if it works for you.

ShodanOne, 2017-07-24:
Hi Alejandro,
Thank you for your quick answer and for the workaround.
It worked also for me.