BCN3D/BCN3DSigma-Firmware #140: Sigma duplication mode disabled ?!?

lemmans, 2017-07-28:
Hi,

In last beta we found, Sigma duplication mode disabled ?!?

This means that standard sigma will not get duplication and mirror modes??

Wich printers are BCN3D_SIGMAX_PRINTER? and wich are BCN3D_SIGMA_PRINTER?

In case people who use this feature (and apreciate it because it saves half of the time printing small parts) we must stay with the last beta that has this option activated?

Other option is generate a new branch and/or compile it by ourselves henceforth modifiyng this new lines of code.

Thanks

AlejandroGarcia92, 2017-07-31:
We are constantly developing new features and new things that could help us to develop our future product strategies.

By now, this feature is disabled because this feature is immature yet.

According to "Other option is generate a new branch and/or compile it by ourselves henceforth modifiyng this new lines of code." is a possibility but not soon.

closed 2017-07-31 by AlejandroGarcia92

lemmans, 2017-07-31:
We have tested it for at least 6 different components, and its working flawlessly for us, with last versions.

In any case we understand its a pure commercial startegy on this decission and we will stay with this beta unitil re-anabled or try to recompile new versions modififyng this part of the code to allow us to work.

In any case great work with this new feature. Its a shame that money rules the world ; )