BCN3D/BCN3DSigma-Firmware #141: Few code bugs (?) - Warnings which dont look right anyway

WanaGo, 2017-08-12:
@AlejandroGarcia92 

Can you please check these lines in the code, as they dont look right, and the IDE is flagging them as warnings, they dont look right. Two of them are memory leaks/out of bounds indexing, of sorts I believe.

-----------------------------

in Configuration.cpp

`#define NUM_LINES 10`

in Marlin_main.cpp, for the gcode_G40()

`float mm_second_extruder[NUM_LINES];
	float i =  -0.5;
	mm_second_extruder[0] = i;
	//for (int count = 1; count <= NUM_LINES; count++){
  for (int count = 1; count < NUM_LINES; count++){
		mm_second_extruder[count] =  mm_second_extruder[count-1] + 0.1;
	}
	//float mm_second_extruder[9] = {19.6, 19.7, 19.8, 19.9, 20 ,20.1 ,20.2, 20.3, 20.4};`

The comment below, looks fine but what the code above is doing is different. Aside from that, its running from index 1 to 10, but the array is only 10 positions (0-9)
The for loop is generating this warning in the compiler:

> warning: iteration 9 invokes undefined behavior [-Waggressive-loop-optimizations]

to me, it should be `for (int count = 1; count < NUM_LINES; count++){`

That being <, not <=

It would then be making an array in the loop with: 
[0] = -0.5 <--- from before the loop
[1] = -0.4
[2] = -0.3 
[3] = -0.2
[4] = -0.1
[5] = 0
[6] = 0.1
[7] = 0.2
[8] = 0.3, 
[9] = 0.4

I believe

It does the same later, in gcode_G41()

With it out it is, its writing to index 10 of the array, which doesn't exist, so this is a memory leak I believe.

------------------------------

In termperature.cpp there are a few instances of this (and other sorts):

`#ifdef THERMAL_RUNAWAY_PROTECTION_PERIOD && THERMAL_RUNAWAY_PROTECTION_PERIOD > 0` 
which bring up warnings 

> warning: extra tokens at end of #ifdef directive
Pointing at the &&.

This doesnt seem right to me, to be in one line. I am unsure if this is working or not.
Should this not be 2 lines, with 2 #endif to finish them?

`#ifdef THERMAL_RUNAWAY_PROTECTION_PERIOD && THERMAL_RUNAWAY_PROTECTION_PERIOD > 0
		if(thermal_runaway_reset_hotend_state) thermal_runaway_protection(&thermal_runaway_state_machine[e], &thermal_runaway_timer[e], current_temperature[e], 0, e, THERMAL_RUNAWAY_PROTECTION_PERIOD, THERMAL_RUNAWAY_PROTECTION_HYSTERESIS);
		thermal_runaway_protection(&thermal_runaway_state_machine[e], &thermal_runaway_timer[e], current_temperature[e], target_temperature[e], e, THERMAL_RUNAWAY_PROTECTION_PERIOD, THERMAL_RUNAWAY_PROTECTION_HYSTERESIS);
	#endif`

to:

`#ifdef THERMAL_RUNAWAY_PROTECTION_PERIOD 
    #if THERMAL_RUNAWAY_PROTECTION_PERIOD > 0
		  if(thermal_runaway_reset_hotend_state) thermal_runaway_protection(&thermal_runaway_state_machine[e], &thermal_runaway_timer[e], current_temperature[e], 0, e, THERMAL_RUNAWAY_PROTECTION_PERIOD, THERMAL_RUNAWAY_PROTECTION_HYSTERESIS);
		  thermal_runaway_protection(&thermal_runaway_state_machine[e], &thermal_runaway_timer[e], current_temperature[e], target_temperature[e], e, THERMAL_RUNAWAY_PROTECTION_PERIOD, THERMAL_RUNAWAY_PROTECTION_HYSTERESIS);
	  #endif
	#endif`

There are a few instances of this.

----------------------------------------------

In Touch_Screen_Definitions.h

`#define BUTTON_SDLIST_SELECT5				847`

There is the warning 

> warning: case label value exceeds maximum value for type

847 is certinaly not valid, but I cannot see this used in the Workshop application anyway. Maybe 847 was just added for a place holder. Either way, it should be corrected or removed, shouldnt it?
the SDLIST is from 0-4 from what I can tell, there is no 5

---------------------------------------

In the ConfigurationStore.cpp file, at the end of the EEPROM_READ_VAR listing, before UpdatePID();, the #endif from the previous section ends with this

`EEPROM_READ_VAR(i,bed_offset_version);
#endif RECOVERY_PRINT`

This is the only instance like this, but the compiler is complaining about the 'RECOVERY_PRINT' after the #endif 
This should be removed I believe.

Regards

AlejandroGarcia92, 2017-09-04:
Thank you, I will take a look soon.

AlejandroGarcia92, 2017-09-13:
First, I changed to:
`float mm_second_extruder[NUM_LINES] = {-0.5, -0.4, -0.3, -0.2, -0.1 ,0 ,0.1, 0.2, 0.3, 0.4};`

I don't changed that because it comes from marlin and works fine:
```
#ifdef THERMAL_RUNAWAY_PROTECTION_PERIOD && THERMAL_RUNAWAY_PROTECTION_PERIOD > 0
		if(thermal_runaway_reset_hotend_state) thermal_runaway_protection(&thermal_runaway_state_machine[e], &thermal_runaway_timer[e], current_temperature[e], 0, e, THERMAL_RUNAWAY_PROTECTION_PERIOD, THERMAL_RUNAWAY_PROTECTION_HYSTERESIS);
		thermal_runaway_protection(&thermal_runaway_state_machine[e], &thermal_runaway_timer[e], current_temperature[e], target_temperature[e], e, THERMAL_RUNAWAY_PROTECTION_PERIOD, THERMAL_RUNAWAY_PROTECTION_HYSTERESIS);
	#endif
```

I removed :` #define BUTTON_SDLIST_SELECT5 847`

and I removed `#endif RECOVERY_PRINT`

WanaGo, 2017-09-13:
Thank you Alejandro.

What version of Marlin is the Sigma Marlin based on?
It seems like it is quite old, or am I mistaken?

AlejandroGarcia92, 2017-09-18:
I can't really say from what version we did the fork because we did it a long time ago. So, it's a quite old.
https://github.com/MarlinFirmware/Marlin

closed 2017-09-19 by AlejandroGarcia92