BCN3D/BCN3DSigma-Firmware #142: G36 calibration seems to assume temperatures from last Load

WanaGo, 2017-08-13:
Hi,
I have not looked this up in code yet, but I had PLA on the left, and just loaded ABS on the right. I then did a G36 calibration, and it took ages to heat things up. I went through the process and at the end of it, immediately jumped to the Temp menu to see current temperatures. Extruders were BOTH on, at 240+ degrees, and the bed was 85+ degrees. So these are ABS temperatures, even though it only uses the LEFT extruder for G36, and in the left is mounted PLA, not ABS.

Also there is more problems, as there is no way to tell the printer what size Nozzle you have, let alone the material that is loaded. So what settings does it actually use for calibrations? Does it always assume a 0.4mm nozzle? Does it always assume the last material loaded is the one loaded into both extruders, and that is what is used for Calibration?

Think some work needs to happen here, as you get very different results when trying to calibrate than you do in a real print, if you are not using 0.4mm nozzle.

AlejandroGarcia92, 2017-09-04:
I will take a look on that issue. Actually, Calibration and G36 also are implemented for 0.4mm size nozzle. By now, there is no way to set the nozzle size by lcd screen so I would do something that needs to be set via USB commands.

WanaGo, 2017-09-13:
USB commands would be fine, as I can edit the display software myself to add in display control, if there is matching commands in the Marlin firmware which I can trigger. Thanks