BCN3D/BCN3DSigma-Firmware #143: Pause Print

4truemodels, 2017-08-17:
Hi there!
I love the "Pause Print" option of the Sigma, it lets me change the filament safely and resume the print afterwards. I can't recall if that order keeps the hotends hot or not, if that is the case maybe it should be better to turn them off until new order, like change filament or resume print when the printer should heat them again, purge and keep running. I would only keep the heatedbed hot just in case the user needs to keep the printer paused too many time so some filaments can cause troubles into the hotends just heated and stopped in there.

My question, or suggestion, is: what specific order is computed to the Sigma's firmware to start the "pause print" protocol? I'm asking that in order to introduce that order in the Gcodes through scripts or plugins for both Cura and Simplify3D. 
This way I should be able to program a "pause print" at a certain height so I can change the filament for different colors or materials. Combining smart designs, both extruders and this feature would be lovely and let the users print almost without limits! 
Therefore, if we can settle that script or plugin in order to make the Sigma "beep" to let the user know when it get paused waiting new orders, that would be great!

Thank you to all the BCN3D team for the good job!

AlejandroGarcia92, 2017-09-04:
Hi @4truemodels, I will give you an answer next week hopefully.

AlejandroGarcia92, 2017-09-19:
It could be nice. I think that there is a way to implement this feature, not the Sigma "beep" but something can be done. But right now we are not focused on it.

norbertut, 2017-09-21:
As of now I only found the "Pause at height" plugin in Cura, it inserts and M0 gcode command at the height you specify. but it did not manage to make it work with the SD.

I managed to make it work with the SD in Simplify3D running the print with an USB connection. I guess Cura also can work if connected to the printer.

Desperate for having this M0 or M600 command to work in standalone prints!!!

4truemodels, 2017-10-05:
Hey there!
Still no advance to being able to incorpore this option?
I would love to know which orders I may introduce in a Simplify3D script so I can control the "Pause Print" in a specified layer and let the printer do it automatically. 
I thought it should be really easy to introduce some order in a script that has the same effect than really pressing the "Pause Print" option in the screen while printing. Please put an eye of that, would be really awesome to have this feature not only "manually" as it goes now but also "automatically" introducing it into the gcode someway.

norbertut, 2017-10-06:
You can do this in Simplify3D, but it only works with the Sigma connected through USB. I opened an issue as improvement to have it also work with SD card.

In the Script tab add this line in the post processing piece:
```
;Filament change code below. Specify layer to swap filament. Print will pause and you can change filament, resume print in Simplify3D
{REPLACE "\n; layer 33" "\nG1 X10 Y5\nM0\n; layer 33"}; pause before layer 33, move to X10 Y5
```
This moves the extruder to position X10 Y5 before layer 33 and then pauses with M0 command
You can change the layer number to your needs by checking the layer in your print preview.
And you can also add other commands.
This is a replacement of the comment line starting with ; and to ensure it only picks the comment line we add \n .
Works for me, although I prefer a standalone option.

Starting with this thread you can find more details on this: https://forum.simplify3d.com/viewtopic.php?f=8&t=4231