BCN3D/BCN3DSigma-Firmware #144: Surfing 0 not set after G36 Calibration

WanaGo, 2017-08-19:
After G36 is completed, I noticed that going into the Temperature menu, the values were incorrect and static. Exiting that screen, to go back to the Main Menu, and then going back into Temperature menu again, and the values were correct again and moving.
This is also noticed if you do a print after a G36, the temperatures in the print menu can be static.

I connected Pronterface, and saw that after the calibration was completed, there was no mention of Surfing 0. Going into Temperature Menu, still no mention. Exiting Temperature menu, and Pronterface showed "Surfing 0" in the log. Going back into Temperature and the temperatures work again.

Looking in Malin:

`case FORM_TEMP:
				if (millis() >= waitPeriod_button_press){
					
					**surfing_temps = true;**
					waitPeriod_button_press=millis()+WAITPERIOD_PRESS_BUTTON;
				}
				break;`

surfing_temps is set true, and the code requires it to be set true here it seems

and here

`case BUTTON_MAIN_TEMPS:
					if (millis() >= waitPeriod_button_press){
						genie.WriteObject(GENIE_OBJ_FORM, FORM_TEMP, 0);
						HeaterCooldownInactivity(false);
						int tHotend=target_temperature[0];
						int tHotend1=target_temperature[1];
						int tBed=target_temperature_bed;
						
						
						flag_temp_gifhotent0=false;
						flag_temp_gifhotent1 = false;
						flag_temp_gifbed = false;
						**surfing_temps = true;**
						waitPeriod_button_press=millis()+WAITPERIOD_PRESS_BUTTON;
					}
					break;`

`void update_screen_noprinting(){
	static uint32_t waitPeriodno = millis();
	static uint32_t waitPeriod_p = millis();
	static int8_t processing_state = 0;
	**if (surfing_temps){**
		//static uint32_t waitPeriod = millis();
		if (millis() >= waitPeriodno)
		{
			int tHotend=int(degHotend(0));
			int tHotend1=int(degHotend(1));
			int tBed=int(degBed() + 0.5);
			char buffer[25];
			memset(buffer, '\0', sizeof(buffer) );
			//Edit for final TouchScreen
			
			sprintf_P(buffer, PSTR("%3d%cC / %3d%cC"),tHotend,0x00B0,(int)degTargetHotend0(),0x00B0);
			genie.WriteStr(STRING_TEMP_LEXTR,buffer);
			
			sprintf_P(buffer, PSTR("%3d%cC / %3d%cC"),tHotend1,0x00B0,(int)degTargetHotend1(),0x00B0);
			genie.WriteStr(STRING_TEMP_REXTR,buffer);`

I am unsure if this code is related to how this works, as there is very little commenting in the code to figure out what 'surfing_temps' actually means. But there is a problem here, if you are able to have a look and find what is the cause.

Running 1.2.5 Release

Thank you.

AlejandroGarcia92, 2017-09-04:
Thank you for your analysis, I will give you an answer next week hopefully.

AlejandroGarcia92, 2017-09-19:
Yes, I have to do something to fix it. I will report you what I change by giving you the commit link.

AlejandroGarcia92, 2017-09-21:
I believe that it could be enough https://github.com/BCN3D/BCN3DSigma-Firmware/commit/2fbe15178a0b9debbac2a89d6dbe85612a96c537