BCN3D/BCN3DSigma-Firmware #146: Start of Print, warmup, Why hotend starts heating so early

WanaGo, 2017-08-24:
Hello,

Printing ABS.
Bed is set to 100 degrees. It takes a long time for the bed to reach this temperature.
But the hotend starts warming up at 60 degrees C, so its sitting there warmed up to 240 degrees, for a long time, while it waits for the bed to reach 100 degrees C.
I am just wondering why this is, and if it can be changed so the hot end doesn't start warming up until the bed has reached its temperature setpoint.

What I am finding is the hotend is sitting there hot for a long time, and all the filament that was inside, has oozed out, making a big mess all over the nozzle, which often does not get wiped away when the print starts, and it drags it over the bed. Also the filament inside is too hot, as its just been sitting in the hotend. It does its initial prime, and often this is not enough, due to all the filament which has oozed out.

Can this be changed simply so the hotend does not heat up until the bed has reached temperature, as the hotend is very fast to heat up in comparison.

guillemavila, 2017-09-13:
Hi!

If you are using ProGen profiles this is normal. On the Start GCode section, remove the lines starting with M190, M104, M109. This will reverse to the default heating sequence of S3D/Cura. That is, heat the bed first and then the hotends.

In ProGen we designed a custom heating sequence where the bed and hotends reach its own target temperature almost at the same time, according to our tests. This reduces waiting time before printing.
However, all components have variability and in specific scenarios it can lead to the situation you comment when bed heats slowly but hotends heat fast. In these cases we recommend to disable the custom heating sequence.

WanaGo, 2017-09-13:
Thanks Guillemavila
Surely there cannot be that much variability in parts...
How long does it take from room temperature to 100C on your printer there? Mine quite literally takes 30-40 minutes. It reaches 60C in <10 minutes, so its sitting for about 30 minutes at full hotend temperature, waiting for the bed to heat up.

This is a real problem, especially with the 2nd (right) extruder for some reason, as it never starts printing correctly when the print starts, its always printing nothing to start with, even though the prime can release some filament before it starts. I am having to bring Skirts, and it takes until the 3rd skirt line for filament to start coming out.

guillemavila, 2017-09-19:
We are aware of the variability of bed resistances. Most of them take around 10-15min to reach 100C. If in your case the bed is taking way more than 15min and it's a problem to your day to day use of the machine I encourage you to contact our technical support and for sure you will find a solution to this issue.

Quick workaround by now is to disable the custom sequence and heat up first the bed, second the hotends.

What slicing software are you using? If the hotend starts underextruding is usually more related to extrusion than temperature.

WanaGo, 2017-09-30:
After lots of testing, it seems to have nothing to do with temperature or extrusion temperatures. I have set up a Poll on the Facebook Sigma Lovers and Owners group, and I am not alone with the problem relating to the right extruder, and it failing to start printing right away, even after purging successfully before it starts. There seems to be an issue here. It occurs with a dual print, left extruder starts fine, right extruder will not extrude for the start of the print, which is evident when you do a skirt. I am printing a large part currently, and each time it starts it takes till the 2nd skirt line for the filament to start coming out. After that, no problem.

Relating to my bed, it still takes 30+ minutes to reach 100C. I will contact support to see what they say about this. Once it reaches temperature, it is stable however and manages to stay there OK, but the process to heat it up is extremely slow.

When using S3D, the order of heat up makes the bed heat, then when it reaches 60C the hotends heat up. In Cura, with Vitamins, this is solved and the hot ends do not heat up until the bed has reached its set point - which I believe to be the correct behavour. So why dos S3D not do this? Is that due to it using default settings from the Firmware, unlike Cura which may be overwriting those settings?

Can the firmware be fixed so this is solved?
What about the right extruder not printing at the start on a dual print?

guillemavila, 2017-10-19:
> After lots of testing, it seems to have nothing to do with temperature or extrusion temperatures. I have set up a Poll on the Facebook Sigma Lovers and Owners group, and I am not alone with the [...]

During the first purge at startup, we ended retracting the filament so it was not oozing in the pure tray. Afterwards when the hotend came back to print it had to prime the filament to print properly. 
In Cura we can postprocess the gcode so it primes properly when T1 comes in. With S3D it's not possible as its postprocessing options are too limited.
ProGen 2.0.2: I changed the behavior and now avoids the retract after T1 start purge. It can ooze in the purge tray while T0 starts printing, so my recommendation is to use a long enough skirt to prime T1 at first.

> When using S3D, the order of heat up makes the bed heat, then when it reaches 60C the hotends heat up. In Cura, with Vitamins, this is solved and the hot ends do not heat up until the bed has [...]

We don't implement that feature with Cura 2.
ProGen 2.0.2: Disabled for S3D & Cura. It sometimes can save a minute or two, but the user experience during this heat up sequence is worse as the users don't understand the behavior and too many times it's not as efficient as it should. It's a GCode issue, nothing firmware related.

closed 2020-07-21 by AlejandroGarcia92