BCN3D/BCN3DSigma-Firmware #147: Air Printing - despite calibration

WanaGo, 2017-08-31:
Have you seen this problem before, where the printer will air print, despite doing a full calibration, bed calibration and G36 calibration.... it still just prints in the air...

During the calibration, it prints perfectly on the bed. But during a print, it just goes and prints in the air...

Is the board suffering from EEPROM wear or something, given the application writes to EEPROM so often?

lampi87, 2017-09-11:
Same for me
I've reverted version to 1.2.3 and changed generated profile (progen) to an old one. Now it seems to work fine.

AlejandroGarcia92, 2017-09-12:
Hi, we don't have these issues. I don't know if this issue can be produced by profile, firmware or mechanical causes. I remember that EEPROM can be write/erase up to 100000 cycles.

WanaGo, 2017-09-12:
Hi Alejandro,

Sorry I forgot to update this issue.
The issue ended up being Simplify3D, when the Z-Hop / Retraction Vertical Lift is set to 0, and you have a Skirt, it prints the skirt in the air, which is what I was seeing at the start of the print. I have already reported this to S3D, however by the sounds of it, it will never be fixed.
[image]

When you do dual colour, and have the same setting on 0, it goes even more crazy.
[image]

This was resolved by setting the distance to 0.01mm instead of 0mm.

Other than these, I have not seen printing in the air on 1.2.5, however I have heard of others who have, and mentioned it takes a number of bed level calibrations to get the bed in the right place again. However, this was not my issue.

closed 2017-09-12 by WanaGo