BCN3D/BCN3DSigma-Firmware #148: lockup on filament unload

jp7189, 2017-08-31:
The filament unload procedure works as expected (filament unloads all the way), but the LCD screen gets stuck at "processing". It takes a power cycle of the printer to getting it working again.  This was happening on 1.2.2RC, and is still happening on 1.2.5. 

More info:
I'm using bondtech extruders, and esteps are set to 492.45 on the eeprom - and I fully understand if this mod causes this bug, and am willing to live that consequence.
Also, I have the USB cable unplugged just to rule out interaction with a host.

AlejandroGarcia92, 2017-09-19:
I have more report of stuck at "processing" not only on unload/load filament, also in others stages. I'm working to fix this issue. Right now, I think that I've already fixed this issue on my dev branch but I still doing more testing.

RhepTheSpring, 2017-09-21:
Jp7189
who said it was you had Bondtechs fitted, I had this before Bondtechs

tsp666, 2017-10-14:
I can second that, just installed bondtech extruders and unload filament is stuck, so a change within a print is not possible anymore. NEVER had that problem before the bondtech upgrade. (updated stepper drivers to r17)

tsp666, 2017-10-15:
@AlejandroGarcia92

I made some tests with different firmwares.

on 1.24  -> unload stuck, loading ok
on 1.25  -> unload stuck, loading ok
on 1.26  -> unload stuck AND load stuck too! 

so it seems 1.26 made the problem worse ?

This is on a upgraded sigma to R17 with just new bondtech extruders (and eeprom value corrected)

So i am downgrading to 1.25 again for the time being.
With standard extruders it always worked well, also pausing during print and changing filament.

AlejandroGarcia92, 2017-10-16:
@tsp666 According to processing stuck issue, I talk about it here https://github.com/BCN3D/BCN3DSigma-Firmware/issues/136 . I hope that with the new LCD screen library, marlin would be able to reestablish the communication with screen as I tested. 
I'm still investigating why a timeout takes place but it seems that stepper affects on the communication with LCD causing corrupts bytes(to sum up).

tsp666, 2017-10-16:
@AlejandroGarcia92  For me it always worked with the original extruder steppers, just since the new bondtech ones this error happens. Many thanks!

closed 2018-10-03 by AlejandroGarcia92