BCN3D/BCN3DSigma-Firmware #152: LCD Screen image upside down after the firmware update v01-1.2.5

calvinkim207, 2017-09-19:
Hello,
I've just received a new BCN3D Sigma R17 and did a Firmware update to v01-1.2.5 (latest).
It was a simple exercise to delete the old files in the card and copy the downloaded files.
When I loaded the SD card back, the firmware reads and works fine except the image is upside down.

How can I fix this??

AlejandroGarcia92, 2017-09-19:
You have installed SD files of R16. We have not uploaded the R17's SD files on Github because we don't have in mind to change them by now.

I let you here the R17's SD files

[SD files v122-LCD-R17.zip](https://github.com/BCN3D/BCN3DSigma-Firmware/files/1313887/SD.files.v122-LCD-R17.zip)

closed 2017-09-19 by AlejandroGarcia92

calvinkim207, 2017-09-20:
Thank you very much.
It is now solved!

AlejandroGarcia92, 2019-08-28:
@svenedt Do you need help?