BCN3D/BCN3DSigma-Firmware #153: Temp issues and temp overshooting after firmware upgrade to v01-1.2.5

Prejkox, 2017-09-27:
Hi,
After upgrading to v01-1.2.5 the printer have started to behave strange when it comes to the temperature. It is the same for both hot ends. 

Issue no1: When staring a print it has stopped to wait for the heated bed to first reach its temperature before heating the hot end. It begin to heat the hot end at once when pressing start print. (Before upgrading it waited for the heated bed to reach its temperature before heating hot end)

Issue no2: It overshoots the the temp setting done in the slicer (I use S3D) by about 30 degrees for about a minute or so before stabilizing on the preset temperature done in slicer. This never happen before with the old firmware (01-1.2.4).

I have checked the slicer and checked if it is me who is the problem, but I have not found that it is any setting that I have done that causes these two above issues. I have mark/unmarked the option "wait for the temperature controller to stabilize before beginning build" but the printer behave the same anyhow. All profiles I use if from the profile generator on BCN3D homepage.

Best regards Tobias

Prejkox, 2017-09-28:
I continued to try to understand what this issue was late last night after I wrote above message.
I tried even more profiles and then the temperature was suddenly stable again as it should and it was waiting for the heated bed to get warm. All the profiles I tried to that point caused the strange behavior i mentioned in previous post. I don't understand this.... I compared the profiles but could not find any difference between them. I have not done any change in the profiles what I can see that should cause this behavior. 
What I'm trying to say is that it may be slicer profile related but I don't understand how and why. 
The thing is that I upgraded the firmware some days ago and directly after this I found this issue. The profiles I used when seeing this I used before firmware upgrade with no issue. That was why I got to the conclusion that it may be firmware related.

Best regards Tobias

guillemavila, 2017-09-28:
Hi Tobias,

as you say it looks clearly related to the profiles used with Simplify. If you have both files you can easily check changes [here](https://www.diffchecker.com).
Note that yesterday we updated ProGen to v2.0, so maybe a new profile you imported had some conflict with an older one... 
Anyways, it looks strange and we never experienced it. If everything is working now I'll close the issue.

Thanks for reporting!

Guillem

closed 2018-01-30 by AlejandroGarcia92