BCN3D/BCN3DSigma-Firmware #154: Change in the process when changing filament

Prejkox, 2017-09-27:
Hi,

Something that have bothered me since I begun to use the printer is that after a filament change the hot end by default is turned on (have not experienced this on any other printer). This until it is turned off manually thru the temp menu. I have several times forgot this and the hot end has at that time been on for a very long time like an hour or so. I see a big risk with this and would have it turned of by the machine automatically and directly after the filament is loaded. For example when the sign "success" appears.
I see far more issues with having it as it is (turned on) than having it automatically shut of after filament change is done. The biggest issue as I see it is that the hot end is getting empty after a short wile due to the heat and I need to clean it more often due to burned filament in the nozzle because I forgot it an hour or two when doing something else. It also adding some points on the dangerous scale when you think that the hot end is off but it is not.

Please fix this "issue" or add a option in some menu so you can choose if you want it on or off after changing filament if this behavior is useful to any user. If there is any reason for having it like it is, please explain it to me. :-)

Best regards Tobias

AlejandroGarcia92, 2017-09-28:
I understand, that's why the printer turn off the hotends after 10 min of inactivity while surfing in the menus but maybe it's not enough.

Thanks Tobias

Prejkox, 2017-09-28:
Ok, then my perception of time may be the biggest issue here, sorry for that. I have never seen it been turned of automatically and have always needed to turn it off manually and I really thaugh that at least an hour had past by at several times i forgot it. I understand now with your info that this is not the case.
Anyhow If it is like that, that it turns off after 10min then it much less serious than I wrote.
Still Is there any reason for why it is turned on for 10min? Only reason I can find is that I maybe save like 1min if I start a print at once.
Still i would suggest if possible to turn it off directly after filament have been loaded if there is no special reason or very important to other users to have it on for that extra 10min.

Best regards Tobias

AlejandroGarcia92, 2017-10-04:
@Prejkox We will take into account your suggestions. I will keep this issue opened until i implement something related with that.

Thank you.