BCN3D/BCN3DSigma-Firmware #157: Pause at layer with M0 - USB works, SD not

norbertut, 2017-09-29:
I have set up a script in Simplify 3D to insert a pause command M0 at a defined layer, plus and X and Y move after that.

The pause is executed when I launch the print from Simplify 3D with the Sigma connected to the computer via USB.
But if I run the same gcode from an SD card, no pause is performed. IT is reading the commands properly as it executes the XY move inserted after the pause, but there is no stop.

Not sure if this is something firmware related, or I should be using another command for pause. I also tried M600 with the same result.

AlejandroGarcia92, 2017-10-02:
These commands are not enabled in our marlin.

norbertut, 2017-10-04:
El podem passar a petició de millora en comptes de tancar-lo?

Gràcies.

*Norbert Rovira*
Twitter: @norbertUT <http://twitter.com/#%21/norbertUT>
norbertrovira.com main blog for Tech thoughts and passions
sisifoescalador.eu for mountain, running and personal life

2017-10-04 14:20 GMT+02:00 Alejandro Garcia Moreno <notifications@github.com

AlejandroGarcia92, 2017-10-05:
@norbertut Exactament vols un comando perquè s'aturi i vagi al pàrquing? Si es aquest el cas, crec que no hi ha un gcode que ho faci ara per ara. Sento haver tancat l'issue.

norbertut, 2017-10-05:
Però pel que trobo a Internet M0 o M600 sí actuen amb un Marlin des de LCD,
sense connexió USB. Amb la Sigma no és el cas, només via USB

Vols dir que al vostre Firmware no hi ha res?

Per exemple el M600 fa pausa per canvi de filament
http://marlinfw.org/docs/gcode/M600.html

i un exemple d'algú que l'està fent servir amb Marlin i LCD
https://groups.google.com/forum/#!topic/asrob-uc3m-impresoras-3d/hQ8twmM4nEA

*Norbert Rovira*
Twitter: @norbertUT <http://twitter.com/#%21/norbertUT>
norbertrovira.com main blog for Tech thoughts and passions
sisifoescalador.eu for mountain, running and personal life

:

AlejandroGarcia92, 2017-10-05:
El M600 sí està però està desactivat perquè està funcionant s'atura Un altre LCD. Estudiaré com puc implementar-per a la propera _release_ perquè funcioni per a la nostra pantalla. Quan tregui una versió beta, t'aviso.

IdiocyCubed, 2017-10-14:
This is a work around which is not an immediate stop! however for the purposes of pausing for layer change this is absolutely fine.  The code below sets the printpause flag which then pauses the print in the same way that pressing pause on the printer touchscreen does...  I'm not sure why M25 which is pause print is not implemented in the same way???  M25 could be changed to as per below as well.  
```
inline void gcode_M0_M1(){
	#if defined(SIGMA_TOUCH_SCREEN) && defined(SDSUPPORT)
			// If we are SD printing, pause the print by setting this flag:
		if (card.sdprinting) {
			flag_sdprinting_printpause = true;
		}
	#endif
}
```
My following of the pause process from the touch screen is this...
1.) When you press pause on the display it sets `flag_sdprinting_printpause = true`
2.) Then in the update screen printing function it executes this if the above flag is set:
```
                        card.pauseSDPrint();
			SERIAL_PROTOCOLPGM("�PAUSE! \n");
			flag_sdprinting_pausepause = true;
```
3.) Then in get command function if card.isprinting is false, then check if flag_sdprinting_pausepause is set, if it is then they enque G69 and set the processing gif...
4.) Then in G69 it actually pauses the printer moves the hotends out of the way etc. and then sets screen_printing_pause_form = screen_printing_pause_form1
5.) Then finally again in the update screen printing function, if the card is paused and `(screen_printing_pause_form == screen_printing_pause_form1)`  then it sets `screen_printing_pause_form = screen_printing_pause_form2;` then finally it updates the display buttons for stop settings and resume.

This entire process seems much more convoluted than it needs to be and this should be addressed!
@AlejandroGarcia92 @WanaGo