BCN3D/BCN3DSigma-Firmware #162: Possible Bugs and fixes...

IdiocyCubed, 2017-10-09:
Here is a list of things that need some attention.  I have found these going through the code while adding some enhancements.  Line numbers should be correct for 1.2.6 release.  
@AlejandroGarcia92 @WanaGo

in file - lcd_handler.h
function - void myGenieEventHandler(void)
change lines #125 and #126 - 
```
static long waitPeriod_s = millis();
static long waitPeriod_button_press = millis(); 
```
to - 
```
static unsigned long waitPeriod_s = millis();
static unsigned long waitPeriod_button_press = millis(); 
```
Fix - Remote possibility of lockup on overflow

in file - lcd_handler.h
under case label: BUTTON_UTILITIES_MAINTENANCE_NYLONCLEANING_SUCCESS:
Change -
	line #2916: `gif_processing_state == PROCESSING_STOP;`
to - 
	`gif_processing_state = PROCESSING_STOP;`
Fix - Shouldnt be a comparison, it should be an assignment

in file - marlin_main.cpp
in function smartPurge_Distant()
Change -
	line #1122: `double C_eq1 = (C_eq, 1/3.); ` // Possible bug?
to - ?
Fix -	unknown maybe missing pow operator

in file - marlin_main.cpp
in function update_screen_printing()
change -
	line #1152: 
`else if(screen_printing_pause_form == screen_printing_pause_form0 || (screen_printing_pause_form == screen_printing_pause_form2 )&& card.sdispaused){`
to -
 `else if(((screen_printing_pause_form == screen_printing_pause_form0) || (screen_printing_pause_form == screen_printing_pause_form2 ))&& card.sdispaused){`
change -
	line #1947: 
	`if(card.sdprinting && !card.sdispaused || !card.sdprinting && card.sdispaused )`
suggest:
	`if((card.sdprinting && !card.sdispaused) || (!card.sdprinting && card.sdispaused) )`

in file - marlin_main.cpp
in function gcode_G41
change - 
	line #4184:` st_synchronize;`
to - 
`st_synchronize();`
Fix - It is missing the ()

in file - marlin_main.cpp
in function inline void gcode_G43(){
change -
	line #4544: `gif_processing_state == PROCESSING_STOP;`
to -
	`gif_processing_state = PROCESSING_STOP;`
Fix - Shouldnt be a comparison, it should be an assignment

AlejandroGarcia92, 2017-10-09:
@IdiocyCubed Thank you, I will report here the commit.

AlejandroGarcia92, 2017-10-09:
I've made these changes in branch v01-1.2.7dev and I've compiled the code again and these are on /Compiled Firmware too.

AlejandroGarcia92, 2017-10-09:
1711a29bd70518ce4f5212eedcc14fe734eedc94

IdiocyCubed, 2017-10-09:
Are you compiling with the latest version of Arduino now? @AlejandroGarcia92 I see issue #131 referenced here...

AlejandroGarcia92, 2017-10-09:
Now, I am compiling with arduino IDE 1.8.4. To clarify, I work with armelstudio using visualmicro plugin which uses arduino IDE for compiling.
I realize that has been release a new arduino IDE version(1.8.5) maybe I will move to the latest version soon.

AlejandroGarcia92, 2017-10-09:
Also, v01-1.2.6 official release was compiled with old arduino IDE 1.6.8.

IdiocyCubed, 2017-10-09:
Thanks, i'm also using atmelstudio and visualmicro plugin with Arduino 1.6.8, can you please detail which compiler works in the wiki or in the main readme? that way if you start using 1.8.5 we can know?  A lot of people seem to be confused about compilers.  Cheers

closed 2017-10-09 by IdiocyCubed