BCN3D/BCN3DSigma-Firmware #163: [enhancement] Fix for the horrible clunking when purging or loading filament

IdiocyCubed, 2017-10-09:
I have tested this little mod, which sets the previous_millis_cmd when a purge is done.  This prevents the inactivity monitor from killing stepper power immediately after the purge is done.  This will keep the steppers alive for 60 seconds, so repeated filament feeds will not have the "Clunk" of the steppers enabling and disabling.  @AlejandroGarcia92  @WanaGo

In file Marlin_main.cpp
In function:
`void update_screen_printing(){`

In lines:
1368 - `if(flag_utilities_filament_purgeselect0){`
1379 - `if(flag_utilities_filament_purgeselect1){`
1390 - `if(flag_utilities_filament_purgeload){`
1400 - `if(flag_utilities_filament_purgeunload){`
1665 - `if(flag_utilities_filament_purgeselect0){`
1676- `if(flag_utilities_filament_purgeselect1){`
1687 - `if(flag_utilities_filament_purgeload){`
1697 - `if(flag_utilities_filament_purgeunload){`

Insert the following before the function call to plan buffer:
`previous_millis_cmd = millis();`

It should look like:
```

		if(flag_utilities_filament_purgeselect0){
			flag_utilities_filament_purgeselect0 = false;
			if(degHotend(purge_extruder_selected) >= target_temperature[purge_extruder_selected]-PURGE_TEMP_HYSTERESIS){
				gif_processing_state = PROCESSING_PURGE_LOAD;
				current_position[E_AXIS]+=PURGE_DISTANCE_INSERTED;
--->>> This line here --->>>	  previous_millis_cmd = millis();
				plan_buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS], INSERT_SLOW_SPEED/60, purge_extruder_selected);//Purge
				st_synchronize();
				gif_processing_state = PROCESSING_STOP;
				genie.WriteObject(GENIE_OBJ_VIDEO,GIF_UTILITIES_FILAMENT_PURGE,0);
			}
		}
```

WanaGo, 2017-10-09:
Confirmed and working here

AlejandroGarcia92, 2017-10-10:
I will take a look.

IdiocyCubed, 2017-10-10:
You will also need to add this line to Marlin.h so LCD_handler.h will know what previous_millis_cmd is
`extern unsigned long previous_millis_cmd;`

AlejandroGarcia92, 2017-10-16:
Ok