BCN3D/BCN3DSigma-Firmware #164: Insert SD card

hst68, 2017-10-16:
Since the firmware update 1.2.6 I get the message "Insert SD card" on the display when I insert the original bcn3d R17 SD card. Same with all other SD cards. Reset and restart failed.

WanaGo, 2017-10-16:
This screen you mean?
[image]

hst68, 2017-10-16:
yes exactly

WanaGo, 2017-10-16:
I would highly doubt its a firmware things - likely something has happened to your machine, you are best to contact sat@bcn3dtechnologies.com for support.
Have you had the centre panel out of the floor of your machine? Remove it and check the cable to the motherboard from the SD card is in place still, and look at the SD card circuit board, and make sure nothing looks out of place on that end. On my machine it has a 5V and 3.3V selection switch, it should be on 5V. Your machine may not have this, I am not sure if they changed it at any point.

hst68, 2017-10-23:
The situation has been resolved. The problem was a defective SD card reader

AlejandroGarcia92, 2017-10-24:
@hst68 ok

closed 2017-10-24 by AlejandroGarcia92