BCN3D/BCN3DSigma-Firmware #165: Screen still hanging

axiiiofevil, 2017-10-19:
I have updated to the latest firmware (1.2.6) hoping that the printer would stop hanging randomly after filament purges, loads and unloads. It's still doing it....
Printer : Sigma R16 (the older screen).

Some time ago, I had a lot of issues with the 4D systems screens genie library for arduino boards (UART com's issues mainly) which was never resolved. I recall that the library was difficult to debug as it was (IMHO) overly complex and not well documented.
Maybe they haven't fixed this?

WanaGo, 2017-10-19:
The issue with the hanging is related to the ISR for the steppers. Due to how often this ISR is run, its not giving the Serial ISR an opportunity to clear the bytes from the screen. @AlejandroGarcia92 is working on this currently. Its not specifically the genieArduino library at fault.

Side note, you could try the new beta of the genieArduino library if you wanted (https://github.com/4dsystems/ViSi-Genie-Arduino-Library-BETA), some new features and stability is better, however even this doesn't solve the current lockups, due to the Stepper ISR, as its not caused by genieArduino. A solution is coming though. Hopefully for 1.2.7.

axiiiofevil, 2017-10-19:
Any idea on when 1.27. will be available?

AlejandroGarcia92, 2017-10-30:
No idea. But recently, I just pre-released the 1.2.7 as a RC version. https://github.com/BCN3D/BCN3DSigma-Firmware/releases/tag/01-1.2.7RC_M26.10

jacquesleps, 2017-11-23:
your link is broken (404) !
Same problem here, 
freeze when loading filament
freeze when unloading filament
sometimes freeze when browsing SD files.
Please explain how to load/unload filament, i do not really care about stepper ISR

AlejandroGarcia92, 2017-11-24:
Hi, @jacquesleps 
I've just change the link for release link because I sometimes change the version name it breaks the link.
What we try to fix is the eventual stuck on the animations(in general load/unload filament, calib, maintenance etc) but what you are reporting is that always gets stuck on that and always on the same site(load/unload filament). 

> sometimes freeze when browsing SD files

(we are aware that sometimes happens ) 

I can explain how to load or unload filament manually but if you machine would work fine you should be able to do it fine.

Also I can suggest you to reload and format the lcd sdcard but I'm not sure if it is the solution.

Thank you for the feedback anyways.