BCN3D/BCN3DSigma-Firmware #167: Duplication and Mirror the right hand print head breaks duplication just before print finishes

IdiocyCubed, 2017-10-25:
At the end of a print in duplication or mirror mode, the right hand print head stops moving in the x direction just before the end of the print.  It is like the mode has returned to standard when the end of the file is reached; however there is still 1 or 2 seconds worth of movement in the plan buffer.  The result is a blob on the right hand print and a string of filament due to the lack of movement in the x axis for the last few moves.  The front right corner of the white cube shows that there is a small section which didn't finish (sorry for the bad picture).  This happens on every duplicate and mirror print.  The second picture had a final move that was in the x-axis only and it has left a big blob on the white print. @AlejandroGarcia92  @WanaGo

[image]
[image]

AlejandroGarcia92, 2017-10-25:
Ok, I think that is easy to fix it. I will report the commit.

AlejandroGarcia92, 2017-10-25:
Can you share me the end gcode?

IdiocyCubed, 2017-10-25:
Sure, here is the end segment of the gcode... the entire file is 10MB.   This is for the pangolin body above which has the large blob on it.

```
;TYPE:WALL-OUTER
G1 F2400 E1186.1069
G1 F600 X44.628 Y60.233 E1186.2262
G1 X44.722 Y60.394 E1186.22786
G1 X44.619 Y60.603 E1186.22993
G1 X31.252 Y60.603 E1186.34907
G1 X31.149 Y60.394 E1186.35115
G1 X31.243 Y60.233 E1186.35281
M204 S2250
M205 X15 Y15
G0 F12000 X36.243 Y60.233
;TIME_ELAPSED:17046.277526
G1 F2400 E1179.85281
M204 S2000
M205 X20 Y20
M107

M104 S0 T0		;left extruder heater off
M104 S0 T1		;right extruder heater off
M140 S0		;heated bed heater off
G91		;relative positioning
G1 Z+0.5 E-5 Y+10 F12000	;move Z up a bit and retract filament
G28 X0 Y0		;move X/Y to min endstops so the head is out of the way
M84		;steppers off
G90		;absolute positioning

M104 S0
M104 T1 S0
;End of Gcode
```

AlejandroGarcia92, 2017-10-26:
656aedb87b731fddb13ef8c13c935af750e2e6a3

IdiocyCubed, 2017-10-26:
Can you point out what lines of code fix this issue???

AlejandroGarcia92, 2017-10-26:
With this change I only ensure that it doesn't breaks dupli mode until executes G28

4truemodels, 2017-10-27:
Hi there!
How can I use the duplication and mirror modes with a Sigma R16?
What version of the firmware should I use and what else steps and/or settings would I need to do to make that possible?

Thank you!

IdiocyCubed, 2017-10-28:
Firmware 1.2.5 has mirror and duplication enabled, but there are bugs in that release and you must manually shim both hotends to the exact same height from the bed.  There is no documentation, but setting M605 S2 X105.0 will enable duplication mode... Use at your own risk, things like vitamins prevent filament grinding will cause duplication to break and crash the print head.

AlejandroGarcia92, 2017-10-30:
Yes, 1.2.5 has some bugs. And this feature was disabled for the 1.2.6.