BCN3D/BCN3DSigma-Firmware #168: custom filament temperatures not always used

carlf28, 2017-10-29:
Since upgrading the firmware to 1.2.6, my custom filament settings are not always used by the Sigma.  For instance if I specify 150C load/unload/print, the Sigma likes to load it at 245C, making a mess.

I have also noticed that my bed temperatures are not always used when printing from the SD Card.  I typically use either 40C or 70C, and the bed is almost always preset to 85C or 59C.  I now know to quickly manually over ride these settings to the ones I want.

I suspect the firmware, but have also recently installed the R17 upgrade and the latest version of Simplify3D.

AlejandroGarcia92, 2017-10-30:
Hi @carlf28, firmware compares the old filament and the current filament config, and uses the highest temp in other to remove the old filament that could be on the hotend just in case.

Finally, when you are printing, the printer sets the temps given by the gcode.

carlf28, 2017-10-30:
Thank you.  That would make sense for the hotend temperatures.  I have been switching between printing with PETG to printable wax and PVA with a nylon clean in-between.  Since I am removing the old material manually is there a way to force the firmware to accept my lower temperatures?  I suppose I could do a false filament load and wait for the temperature to stabilize and then do the actual filament load.

If I understand correctly then the hotbed temperature is set in Simplify3D since that is where my code is generated, so likely I have an option checked somewhere that is telling the hotbed to go to 85C or 59C first rather than my lower process setting temperature?

closed 2020-07-21 by AlejandroGarcia92