BCN3D/BCN3DSigma-Firmware #169: Bondtech Kit for BCN3D Sigma

hst68, 2017-11-01:
Bondtech Kit for BCN3D Sigma there are still problems with the firmware 1.2.6, especially in the following cases when loading and unloading the filament, since timeout errors cause the printer to crash.

AlejandroGarcia92, 2017-11-02:
https://github.com/BCN3D/BCN3DSigma-Firmware/releases/tag/01-1.2.7RC_M31.10

AlejandroGarcia92, 2017-11-02:
I hope that this issue will be fixed in the next release. If you can test our firmware release candidate version and send us feedback, it could be useful for us.

AlejandroGarcia92, 2017-11-02:
Remember to set Bondtech steps per mm again.

tsp666, 2017-11-06:
@AlejandroGarcia92 

So far with RCM31.10 no problems with timeouts anymore with bondtech extruders.

AlejandroGarcia92, 2017-11-06:
Thank you for the feedback.

maodch, 2018-01-21:
I'm also owning 2 pairs of Bondtech for Sigma R17.
But I faced in some feeding problems. Tried any variation of Steps per mm from  492.45 up To 510.
The extruder snaps back all the time on all 4 Extruders I tested it.
Going back to the stock extruders I haven't it. First I thought it's the nozzle. After replacing to a new one nothing. Changed anything. But finally after gone back to stock, all is working well again.

Not sure, but after going to 1.2.9 and back to 1.2.7 RC01.12 it was terrible. 
Don't know thats really the matter, exepting the display crashes (which I also had almost after every load/unload), or freezings (randomly). So Not sure if it's only the FW, but in the beginning the Bondtech worked well, but don't remember that Firmware - State.

AlejandroGarcia92, 2018-01-22:
Why did you go to 1.2.9(which is a Sigmax FW) and came back to 1.2.7? If you do this, by installing 1.2.9 steps will set to 492.45 and intalling 1.2.7 will set to 152.

maodch, 2018-01-22:
Sorry fully agree, it was a typer, should be 1.2.6 to 1.2.7

AlejandroGarcia92, 2018-01-22:
Ok, then when the FW version is changed the e-steps is set to default values. That's what happens for sure.

maodch, 2018-01-23:
yes, agree. But it's not to do with the E-Steps. 
Careless if set to 492.45 or 495 or 503 or even heigher.