BCN3D/BCN3DSigma-Firmware #170: Latest firmware locks up if the hotends aren't exact same heights in duplication for sigmax (fix)

IdiocyCubed, 2017-11-05:
It appears that there may be no future support for mirror and duplication on the Sigma, but I must continue to emphasize how useful it is and that it SHOULD at least be an officially supported option for advanced users.  Duplication works great on the Sigma, and it isn't terribly difficult to shim the hotends manually.  

Anyway issue is lockup when the hotends have some offset > 0.05 after calibration.  It gets stuck on form 23 waiting for the user to press yes or no to accept the print with distortion.  How has this not been solved yet???  The issue is the same one that I mentioned before.  Everywhere that returns from the lcd_fsm_lcd_input_logic function must clear the lcd_busy flag or it will lock up.  
```

if(card.cardOK)
			{
				//doblocking = true;
				#if BCN3D_SCREEN_VERSION_SETUP == BCN3D_SIGMA_PRINTER_SIGMAX
				if(listsd.check_extract_ensure_duplication_print()){
					if(abs(extruder_offset[Z_AXIS][RIGHT_EXTRUDER]) > RAFT_Z_THRESHOLD){
						genie.WriteObject(GENIE_OBJ_FORM,FORM_RAFT_ADVISE, 0);
						sprintf_P(buffer, PSTR("The first layer printed with the %s Hotend\nwill be distorted by %d.%1d%1d mm"),
						((extruder_offset[Z_AXIS][RIGHT_EXTRUDER] < 0)?"left":"right"),
						(int)abs(extruder_offset[Z_AXIS][RIGHT_EXTRUDER]),(int)(abs(extruder_offset[Z_AXIS][RIGHT_EXTRUDER])*10)%10, (int)(abs(extruder_offset[Z_AXIS][RIGHT_EXTRUDER])*100)%10);
						//((extruder_offset[Z_AXIS][RIGHT_EXTRUDER] < 0)?"right":"left"),
						Serial.println(buffer);
						//sprintf_P(buffer, PSTR("Info: First layer printed with"));
						genie.WriteStr(STRING_RAFT_ADVISE_Z_OFFSET,buffer);
------>> Add This --->>>      lcd_busy = false;
------>>>This return-->>>	return;
						//sprintf_P(buffer, PSTR("Info: First layer printed with %s Hotend will be %d.%1d%1d mm higher. You can avoid this compensation using gauges.\n")
					}
				}
				#endif 

```
I also recommend searching for any other returns or possible situations where this could happen!

Additionally I see that other functions which have the line `if(gif_processing_state == PROCESSING_ERROR)return;` and that are not inside the fsm_input_logic function have been changed to clear lcd_busy.  This is very dangerous because if it ever clears lcd_busy and we are inside the fsm_input_logic function and call touchscreen update then you can have out of control recursion.  These other returns simply return back to parent function and should not have been changed.  Please check this very carefully recursion is dangerous when not used properly!

AlejandroGarcia92, 2017-11-06:
Ok, i will change it. Also, I'm thinking to change  `if(gif_processing_state == PROCESSING_ERROR)return; `to  `if(gif_processing_state == PROCESSING_ERROR){LCD_FSM_input_buton_flag = -1; lcd_busy = false; return;}` .

IdiocyCubed, 2017-11-06:
Well I see that you have already changed `if(gif_processing_state == PROCESSING_ERROR)return;` to `if(gif_processing_state == PROCESSING_ERROR){LCD_FSM_input_buton_flag = -1; lcd_busy = false; return;}` within the fsm_input_logic function which fixes the issue of lockups when we have an error; however you have also changed these when we are NOT inside the fsm_input_logic function which is not a good idea.  So outside of the fsm_input_logic function you should not be changing  `if(gif_processing_state == PROCESSING_ERROR)return;` to `if(gif_processing_state == PROCESSING_ERROR){LCD_FSM_input_buton_flag = -1; lcd_busy = false; return;}`

AlejandroGarcia92, 2017-11-06:
Yes, I will revert it.