BCN3D/BCN3DSigma-Firmware #171: USB Printing impossible

maodch, 2017-11-06:
I already talked to the Sigma Support and they sent me here.

My mather:
I connected the Sigma direct to the mainboard connector with the shortest cable (0.5m).

1. After connecting Sigma the Printer restarts several times after keeping stable.
Cura recognizes the connection and after start printing properly the following problems come up:
- Every radius is jerking, stopping, zittering (you can see and hear it very well), so it is unable to print a radius or some round stuff.
- The Speed is more random instead the one was set in Cura.
- Sometimes the print starts in the air, instead on the plate.
- Sometimes the print temperature lowers to an random value and prints anyway

2. After setting the firmware configuration in Simplify3D (don't wait for "start"-command, and don't buffer) the print starts well.
- There is no jerking like in cura.
- But randomly the printer drives away in any random direction, printing regions there would be nothing to print, comes back and resume the print path.
- Randomly the printer stucks in place, pushes much of plastic out the nozzle and resumes printing.
- Sometimes the printer goes back to home one Extruder (of 2)
- Sometimes one extruder turns off (falls far under minimum temperature) and reheat again during print
- Sometimes the printer reboots and connects again to Simplify

3. Cura and Simplify I also use with Wanhao Duplicator i3, Craftbot 2, Davinci Pro and more, without any of this unusuable behaviours.
For now, printing with USB is really unusuable and for that money I speant I'm really unlucky at the moment, because I don't want to go the old fashioned SD-Card way :-(

The support meant, that might be not the first time, because it happend similar with SD-Card printing formerly. So I should place my matter here.

Thanks in advance

IdiocyCubed, 2017-11-07:
I don't work for BCN FYI.  I have found a major issue with the firmware which causes serial characters to be dropped.  This has been implemented to fix issues where characters are dropped from the display and this was causing the lockups in 1.2.6 as far as I can tell.  I have a very modified version of 1.2.7 and I will try implementing the fix for USB and report what I find. Since the USB communication speed is even faster than the display this is an even bigger issue and I believe that this is responsible for your erratic USB behavior.  I have mentioned this, but no official fix has been implemented at this stage, so I will try for myself.  Printing through the SD card is your best option for now.  The other thing to check is whether Cura and S3D implement CRC checks i'm fairly certain they do; however CRC check does not detect all errors and the number of missing bytes which I have been seeing is so high that CRC would definitely not catch them all.  Doing some googling I found this (https://www.bcn3dtechnologies.com/en/forums/topic/simplify-3d-problems-with-sigma/) I'm not sure if anyone has managed to print over USB reliably on the Sigma.  Looking at the firmware in its current state it may be possible if you are printing very slowly since it will spend less time in the stepper interrupt at low speeds.  If anyone else has experience with this and has come up with a solution that would be greatly appreciated.  It is likely that the length of your USB cable will make no difference since the actual baud rate is quite slow for USB anyway.  When Cura generates a radius it breaks it down to a lot of very small linear moves because they are not generating arc commands, so it will be sending a lot more data during an arc and will likely have a lot of errors which is causing your stuttering. @AlejandroGarcia92 The same serial fix which has been used for the display should also be used for the USB and that should help with printing over USB.  Potentially there are other bugs which are not yet known while printing over USB since it hasn't been well tested.

maodch, 2017-11-07:
Thanks IdiocyCubed for let me believe that's no just a matter I imagine my self :-)
And yes, I'm currently using the 1.2.6. If you find a fix it would be early xmas for me. Because it's really unbelievable for a Printer above 2000 Euro having this "Basic" Problems. Honestly, the rest of the Printer also the quality is awsome, but I like to prototype my sample parts, and that's not possible with SD printing, because the fine adjustments you can make trough usb. 
Anyway, I'ts like you're describing above, it makes absolutly sense why the two Software have differences in execution an jerking behaviour. After deeply looking the gCode I got it as well. 

But in this case there is only one Option...to wait and hope that somebody fix it for usb printing.
And in BCN3D Forum mentioned, the fix for the Serial matter really seems to bi almost the same like my matter. But I'm really wondering, why this Problem didn't come up from other People, because I can't be the only one who have bought a Sigma and faced in this issue. Strange.

However, hope to find a solution soon :-o

(and sorry for my bad English)

AlejandroGarcia92, 2017-11-08:
Hi, @maodch. Can you attach here a sample gcode(or model) which this issue happens? If I fix it, i will update the Pre-release version ASAP.

maodch, 2017-11-08:
@AlejandroGarcia92 , wow generated some extra work :-) 
I'll start testing this evening (sorry, timeshift), do you Need a sample stucking gCode-File as well? Or has it already been considered in the new build?

maodch, 2017-11-08:
For example this:

[monster_pencilpot.zip](https://github.com/BCN3D/BCN3DSigma-Firmware/files/1455173/monster_pencilpot.zip)

But not testet the new Firmware so far

AlejandroGarcia92, 2017-11-09:
Pre release firmware is on 01-1.2.7RC branch.

maodch, 2017-11-09:
@AlejandroGarcia92, wow, after installing it was quiete cool to see it working.
But with Simplify3D I got errors like this during a while:
READ: ok
SENT: G1 X101.681 Y178.840 E3.1766
READ: ok
SENT: M105
SENT: G1 X101.681 Y163.750 E3.3319
READ: ok
SENT: G1 X102.031 Y163.750 E3.3355
READ: Display is ready
READ: Display got disconnected
READ: ok
SENT: M105
READ: ok
SENT: G1 X102.031 Y178.840 E3.4907
READ: ok T:223.8 /220.0 B:64.9 /65.0 T
SENT: G1 X102.381 Y178.840 E3.4943
READ: 
READ: splay is ready
READ: splay got disconnected
READ: 

With Cura, same behaviour as before jerking and zittering arround.
I'll update after testing some different things again.

maodch, 2017-11-09:
So, after testing, it works wery well with Simplify, but suddently he make terrible mess and does what ever:
[error.txt](https://github.com/BCN3D/BCN3DSigma-Firmware/files/1459216/error.txt)

maodch, 2017-11-09:
After tool change the ghetto begins, printing not on the prime tower, faster ever set and closing up on non printed grounds .... 
same gcode, with SD-Print, works fine without that cracy stuff

maodch, 2017-11-09:
The "disconnect" command was from me, because i had to kill the progress.
uploading here the full original
[SupportTest.zip](https://github.com/BCN3D/BCN3DSigma-Firmware/files/1459226/SupportTest.zip)

 gCode:

maodch, 2017-11-09:
and don't ask, why I'm printing that crap in this position, has the reason to testing out new PVA and HIPS :-)

maodch, 2017-11-11:
Hope, there are some more ideas?

AlejandroGarcia92, 2017-11-13:
I don't know what happens with simplify3d but I never use it for usb printing. I used to use Pronterface or Repetier-Host. Now we are working to implement it for Cura2 as well.

AlejandroGarcia92, 2017-11-13:
Also this bug was reported here #77:

```
SENT: M105
SENT: G1 X110.545 Y157.955 E0.2053
READ: ok
SENT: G1 X109.200 Y157.955 E0.2220
READ: ok
READ: ok
SENT: G1 X109.200 Y167.045 E0.3342
READ: ok T:221.
SENT: G1 X108.455 Y167.045 E0.3434
READ: 
READ:  T:.0.B:./.T0./0. T1.0.@B@:
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ: 
READ:  T:1.1 /220.0 B:64.9 /65.0 T0:220.1 /220.0 T1:221.1 /220.0 @:63 B@:53
READ: ok
```

AlejandroGarcia92, 2017-11-13:
To sum up, what I can say, it works well on Pronterface and Repetier-Host  for me.

maodch, 2017-11-16:
Yes sure, but that isn't a solution, because Cura and Simplify are the most used 3D Slicer's.
And as mentioned, I don't understand why other really cheap manufacturers like Wanhao haven't problems getting that running on both. And we're talking about 350$ Printers not 2500$ Printers

maodch, 2017-11-25:
@AlejandroGarcia92 
I also tested Cura with Repetier Host (Octoprint), and same result as printing directly from cura.
The result isn't round, seeded with zitts and blobs, really ugla. 
Same GCODE with SD printed works perfekt.
So, it isn't really fixed so far.

Any further ideas?

You must test it with round things, straight lines aren't the matter.
[image]
[image]

maodch, 2017-11-25:
[image]

AlejandroGarcia92, 2017-11-27:
Yes, we have similar results. We find that there is a problem when is taking a curve where is processing a lot of gcodes. We are working to find a solution.

maodch, 2017-11-27:
Cool :-) 

Btw: is there a future way, to have the Filament Flow accessible in LCD Touch?
Because Thats the only thing left I'm missing, when printing via SD. 
Because temperature, flow and speeds are the 3 most important settings during a print.

AlejandroGarcia92, 2017-12-19:
Hi, we've recently made some fixes on the USB communication and now it's possible to establish connection and print a job in simplify3d. 

I use the following communication firmware config:
[image]

And I got good results. No slow curve moves as a sd printing. Let me know, we are still working on that.

Those changes are in the RC version:  https://github.com/BCN3D/BCN3DSigma-Firmware/releases

maodch, 2017-12-20:
Hi @AlejandroGarcia92 

very cool, Updated and tested. Thank you!

So far I'm with you, no issues. My first part was easier, less roundings, but I will update you soon after my Pingu test :-)

BTW: The Flowrate, is it accessible too trough S3D? Because I took your recommended settings, and there Looks like it isn't?

Off Topic question: Is it also possible to get over 100° (actually 109°) degrees for the build plate, because of hard parts like HT printed with 280°C, should have a heat bed arround 220°-230°. I recogniced a death end at 110°, it shut's down the heater totaly after reaching it.
Any whay, without loosing warranty getting to 220°?

AlejandroGarcia92, 2017-12-20:
BTW: The Flowrate, is it accessible too trough S3D? Because I took your recommended settings, and there Looks like it isn't? 

`I will test this S3D feature to ensure if it works. You recommend me to implement this feature in settings but unfortunately we don't have plans to do it soon.`

Off Topic question: Is it also possible to get over 100° (actually 109°) degrees for the build plate, because of hard parts like HT printed with 280°C, should have a heat bed arround 220°-230°. I recogniced a death end at 110°, it shut's down the heater totaly after reaching it.

`
There is a firmware limit which is 110º. But our heatbed would not reach more than 110º because it don't have enough power. `

Any whay, without loosing warranty getting to 220°? 

`Contact support`

Thank you for the feedback

maodch, 2017-12-20:
Something I recognized: Every 3 seconds (or Close to that) the message appears: "Display is ready" and folowed by "Display got disconnected"

But sofar it's running Forward...

maodch, 2017-12-20:
and the Display is getting unaccessible ... no touching anymore :-)

AlejandroGarcia92, 2017-12-20:
I believe that this mainboard seems that has got crazy.

maodch, 2017-12-21:
So. Finally, the prints are really good.
But something is wrong. As soon you have connected S3D to your Printer, printing some parts (without S3D, only with SD), after some prints, the Printer has totally weird coordinates and is dooing strange things.
Same behaviour as in the past.
And really strange, the Printer only has to be connected with the pc, and not doing Special things.

If I could guess: After connecting the Printer with USB the Display is always on, even the Printer is set OFF. It seams, that there are some variables or Settings stored, which will kept over time, and make some Problems.
If the Printer is in the failture more, you can tun it off (Display still online), and turn on again, set in a new sd, start printing, and the Printer still makes same strange things as before. Only Thing that helps is unplugging USB.

maodch, 2017-12-21:
@AlejandroGarcia92 
About the Flowrate you're right, it works with S3D, perfect. I tested it too. Very cool. Thanks.

If the "Mainboard driving crazy mode" wouldn't happen, the rest seams working as designed, nice, and stable. 
Good Job :-)

maodch, 2017-12-22:
@AlejandroGarcia92 
Btw: Wish you a wonderful christmast time :-)

closed 2017-12-22 by AlejandroGarcia92

maodch, 2017-12-22:
@AlejandroGarcia92 
Will this Feature also work for Cura? Because it's your main Software bundled and offered with BCN Printers.

AlejandroGarcia92, 2017-12-27:
That is what we are developing.