BCN3D/BCN3DSigma-Firmware #172: upgrade of firmware R17

gibe14, 2017-11-10:
Cura does not start process from Mac. From  Windows recognizes the current release (1.2.2) and starts downloading new release (1.2.6). It creates a directory but it is empty. Any further attempt fails because cura says that the release was already downloaded (not true). Downloaded manually file V01.1.2.6.hex. Cura starts saying "reading firmware" and never stops. Any help? Thank you

maodch, 2017-11-10:
Hi there

You can download the Firmware from here. 
Then take the old cura (0.1.5 beta) delivered with Sigma and click tools (I think) and "Update Custom Firmware". 

Not sure if it solves your issue, but I also had to take the old BCN-Cura-Edition to keep it working with Windows at least.

gibe14, 2017-11-10:
Thank you Maodch,
I did like you suggested and I was able to install an update. Unfortunately the update does not solve my original problem (temperature in extruder goes down instead of going up when printing ....)

maodch, 2017-11-11:
@gibe14 
Are you printing with SD or USB?
If you're printing in Cura, disable the "Auto-Temerature"-Feature, because I had this issue with this Setting.
Did you try Simplify? There it isn't any issue with temperature contorl, as long you're printing with SD.

gibe14, 2017-11-12:
Thank you again Maodch. What happened is that the same file, printed from SD card and generated by Simplify, suddenly induced a different behavior of the printer. Before it would start heating up the extruder and the plate at the same time. Now it goes with plate first and then extruder. Absolutely same file!!! For that reason I updated the firmware. If I can ask you a question: in the original SD card the folder firmware was empty. Do you have to copy the 1.2.2 files on that folder?

maodch, 2017-11-12:
To be honest, didn't go this way so far. 
Regurlarly you have to copy all the needed files to sd (unplug it behind your Screen) and overwrite it.
or, go the USB-Update way (I alway took this path).
So, as I know, download the original 1.2.2 and overwrite the SD files, and set it back into the display-port (not the external one).

AlejandroGarcia92, 2017-11-13:
The last screen version was released on v01-1.2.2. Those LCD-sdfiles are the same for newer firmware versions. So If you are using v01-1.2.6 you must use 1.2.2 lcd files.

gibe14, 2017-11-13:
Alejandro, the 1.2.2 lcd files must go on the card INSIDE the firmware directory? My SD card originally did not have any of the 1.2.2 lcd files on.
Thank you for your help

AlejandroGarcia92, 2017-11-14:
We have this tutorial: https://github.com/BCN3D/BCN3D-Cura-Windows/wiki/Updating-the-SD-Files-from-the-LCD-Display

If you download the the release v01-1.2.6 Source code the lcd files will be inside of _**SD files v122 R16**_ folder. Finally, you only have to replace the files.

BurtsBCN3D, 2018-06-30:
Alejandro, I had an issue with the R17 LCD screen as well, showing lines/stripes. Problem started after installing firmware update, running latest software version. 
3DJake< reseller, advised to format the SD from the LCD and reload the SD files from R17. I believe around 18 files in totally, all from the latest firmware release v01-2.7. Now the LCD shows "drive not mounded" and system won't start. Only LCD files on the on the SD.   I work from a MAC so can't use the windows method.
Any advises, system is down for several days. Thanks in advance.

BurtsBCN3D, 2018-06-30:
[image] 
before format 
[image]
after format

AlejandroGarcia92, 2018-07-02:
This kind of issue usually are fixed by support, I recommend you to write a email to them. Anyway, those lines and corrupt stuff is produced because the sd card has been corrupted and a solution is formatting it and copy the files.