BCN3D/BCN3DSigma-Firmware #173: New research increases printing speed and reduces errors.

edanweis, 2017-11-14:
I came across this paper which presents a technique to increase speed and maintain print quality. Just wondering if the BCN3D community is aware of it, or whether it could benefit the BCN3D controllers?

[http://www.sciencedirect.com/science/article/pii/S0957415817301277](http://www.sciencedirect.com/science/article/pii/S0957415817301277)

**Abstract**

> A limited-preview filtered B-spline (FBS) approach for minimizing errors in tracking a desired trajectory is presented. In the full-preview FBS approach, the feedforward control input to a stable linear system, with or without non-minimum phase zeros, is decomposed into B-spline basis functions with unknown coefficients; the basis functions are forward filtered using the (modeled) dynamics of the system, and their coefficients selected to minimize tracking errors of the entire trajectory in one batch. Instead, this paper proposes the use of a receding horizon to recursively compute unknown coefficients that minimize tracking errors for small batches (subsets) of the trajectory at a time, by exploiting the local property of B-splines. This allows optimal control signals to be determined at much lower computational cost compared to full-preview FBS, thus enabling online implementation on real-time controllers. The adverse effects of limited preview on tracking accuracy, relative to full preview, are analyzed, and limited-preview FBS is shown in numerical examples to preserve the versatility of full-preview FBS in tracking systems irrespective of their zero locations. The practicality and effectiveness of the limited-preview FBS approach are demonstrated by employing it for online feedforward compensation of tracking errors caused by structural vibrations of a stepper-motor-driven 3D printer. Alleviation of vibration-induced surface waviness and layer-to-layer registration errors, without sacrificing print speed, are demonstrated.

[image]

**Extract:**

> Commercial desktop 3D printers are designed with light, but flexible, structures and driven with stepper motors in order to reduce their cost, size and weight. Stepper motors are typically open-loop controlled with no feedback measurements to help compensate for un-modeled dynamics. As a result, parts manufactured on such 3D printers suffer from surface waviness (aka. ringing or ghosting) and registration errors, caused by stepper motors skipping counts, due to excessive vibration triggered by the motion of the print head or build platform. Such vibration-induced errors not only mar the aesthetics of 3D printed parts, but often lead to highly distorted and hence scrapped parts. The errors could be mitigated via active vibration control techniques. However, such control techniques require feedback sensors and high-sample-rate real-time control hardware, which are not commercially viable for low-cost desktop 3D printers. In this section, we demonstrate the effectiveness of feedforward control using LPFBS for online compensation of vibration-induced errors of such 3D printers, without sacrificing productivity.

lemmans, 2017-11-21:
Really interesting document

AlejandroGarcia92, 2017-11-21:
We are aware.

IdiocyCubed, 2017-11-21:
I think this is a much easier feature to implement which is in the latest version of Marlin.  It would be great if there was a gcode to set the K factor for various filaments.  This would allow the firmware to know what filament is being used which is a great advantage.  I've put this in because recently the acceleration and jerk has been increased to solve the issues with part corners, but this feature is the best fix for that issue. @AlejandroGarcia92  @WanaGo

[http://marlinfw.org/docs/features/lin_advance.html](url)

"Advantages:
- Better dimensional precision due to reduced bleeding edges.
- Higher printing speeds are possible without any loss of print quality.
- Visible and tangible print quality is increased even at lower printing speeds.
- No longer need high acceleration and jerk values to get sharp edges."

"This feature adds extra load to the CPU (and possibly more wear on the extruder). Use a communication speed of 115200 baud or lower to prevent communication errors and “weird” movements. Make sure your print host software is using line numbers and checksums. (This is disabled by default in Simplify3D.)"
#171 
The new version of Marlin has completely restructured the stepper related interrupt routines and they have fixed some of the serial receive issues through this re-structuring.

thesec, 2017-11-21:
+1 on @IdiocyCubed suggestion on adding the linear advance. Had very good results for me and I was able to increase print speed by about 10 mm/s on another printer.