BCN3D/BCN3DSigma-Firmware #174: Firmware v01-1.2.7RC_M11.23 - Feedback

tsp666, 2017-11-23:
Hi,

During calibration, the temperature figures shown are jumping from (example) 80 to 15 then up again and heating takes ages, sometimes stuck for 20 sec.

Manual heating is working flawlessly.

AlejandroGarcia92, 2017-11-24:
Ok, @tsp666 I will check it but this has been changed on 1.2.6 to reduce the dripping by cooling down the hotends temp when is not printing lines. Also, Do you mean that percentage stuck for 20sec?

Thank you

tsp666, 2017-11-24:
@AlejandroGarcia92 ,

See Video of heating process.
It's a brand new 0.6 mm hotend, left, custom material settings of 220degrees.

It took over 9 Minutes to heat up!
The video is showing all the jumps and display errors (sorry a bit boring, but better than words)

https://player.vimeo.com/video/244330118

Regards
Torsten

AlejandroGarcia92, 2017-11-24:
Ok, It seems that hotend has difficulties to reach the target temp if i'm not wrong. Did you try to do AutoTune at Maintenance to update PID coefficients?

tsp666, 2017-11-24:
@AlejandroGarcia92

Yes, Autotune was done just before the recording. Hotend is brand-new (as the previous one had the same effect) This did not happen with older versions of FW.

Normal heating to 220 deg is just taking 1min15 (before printing or manual though temp section, see attached).
So it has to be the calibration routine causing this - I think.

https://player.vimeo.com/video/244342612

IdiocyCubed, 2017-11-26:
The calibration requires the bed to reach temperature as well and this is where I see the oscillating temperatures too... What is the bed temperature for your custom material?  I am installing a more powerful heatbed to solve this issue :)

tsp666, 2017-11-26:
@IdiocyCubed 

the bed temperature was only set to 40 degrees, so this should not have caused the effect.
In earlier Firmware versions this was never a problem.
What kind of heatbed are you looking to install?

IdiocyCubed, 2017-11-26:
Ok there must be something else going on then...  I have a spare heatbed, so I am just doubling them up for now.  I installed a second power supply to handle the increased bed current.

AlejandroGarcia92, 2017-11-27:
@tsp666 Can you show me how much time bed requires to reach the target temperature?  Because, I remember that I let more margin before until temp target is reached.

tsp666, 2017-11-27:
@AlejandroGarcia92 

Is there any other way of measuring it than manually on the heatbed? I can do this with an infrared thermometer if needed

AlejandroGarcia92, 2017-11-28:
I hope that you can. Take into account that the bed thermistor is just in the middle. Try to compare infrared thermometer results between thermistor measure. If it is a bed cause, support can help you better than me.

tsp666, 2017-11-28:
I will measure and report. As normal printing is working fine, I doubt a bed issue, but let's see...

WanaGo, 2017-12-04:
@AlejandroGarcia92 - This is what I experienced a few months ago too, if you recall the video I sent you in email. Same thing.

tsp666, 2017-12-04:
@AlejandroGarcia92 

I measured the real bed temp compared to what is displayed and it's way off. I ordered a new heatbed, let's see how it changes when it arrives in some time.

tsp666, 2017-12-19:
@AlejandroGarcia92 

Heatbed was replaced with a new one, firmware updated to actual beta (19.12.17) - but strange bed temp display still on. While hotends are showing 12 deg, the bed shows 42 ? (as the old one did, actual temp around 14 degs, measured).
Any chance this is caused by any bad eeprom entry or a bug in the fw?

AlejandroGarcia92, 2017-12-20:
It has no sense. In our machines, we don't have problems about it. We don't really change anything in the firmware that could cause this issue(as far as I know). 

To verify:
1. If you disconnect the bed cable, firmware must show 0 deg.
2. Go back to an official release which works fine for sure.

There is another possible option that implies to **mainboard**. So, recently you've changed the Heatbed and still not working which is suspicious.

I've just told you my thoughts around it, I hope that it could be useful if you have to talk again with support.

Send me feedback about the verifications.

tsp666, 2017-12-22:
@AlejandroGarcia92 

It was the mainboard which caused the strange problem - I replaced it and now the calibration process is working as it should, thanks and have a good X-mas time!

AlejandroGarcia92, 2017-12-27:
Have a good X-mas

closed 2018-10-02 by AlejandroGarcia92