BCN3D/BCN3DSigma-Firmware #175: v01-1.2.6 hangs when calibrating

Prejkox, 2017-11-30:
some into "Full calibration" after changing nozzle size printer get stuck after the third test print (the one to check the alignment of the heads). It finish the print and then the screen is stuck with A picture of the head and the text "Printing a test" and noting more happens. After this I need to cut the power.
I have done it twice now and got the same result. The second time the printer did not even start after I flipped the power switch. The screen was just flashing "could not find driver" over and over again. Then   
 I had to pull the wall plug wait for a moment and turn it back on. Then it started again.

So right now I can not perform a calibration. 
Firmware version v01-1.2.6

AlejandroGarcia92, 2017-12-01:
Hi @Prejkox, we were working on this kind of issues related with animation stucks. Take a look at the [pre-release version](https://github.com/BCN3D/BCN3DSigma-Firmware/release). Your feedback is important for us. I hope that it could help you.

Prejkox, 2017-12-01:
OK, thanks for the reply. Seems to be something wrong with the link you included. It points to something that do not exist it says when clicking it. Is it easy to install this when it is not done automaticly thru Cura?
Also Searched for the update firmware option in the new Cura but it seems not be there?! Automated firmware update is still only possible in the old cura version or?

ZAMHelmets, 2017-12-06:
I took the release candidate from[ here](https://github.com/BCN3D/BCN3DSigma-Firmware/tree/v01-1.2.7RC) as the link above didn't work for me. The LCD still freezes on me during the full calibration process. Steps to reproduce:

Click on Full Calibration from the LCD
Run z-axis calibration on both extruders (though I did it once skipping this step)
Run x-axis calibration
Redo x-axis calibration
The LCD freezes and says "Printing Test" after trying to redo the x-axis calibration the 2nd or 3rd time. This happened two times in a row and haven't had time to try a third. The first time I waited overnight and the LCD screen was still frozen the next day. Without the printer calibrated, I'm unable to run any additional prints to know if this is a more general issue. HTH.

AlejandroGarcia92, 2017-12-11:
@Prejkox link fixed. We are working to be able to update the firmware in Cura2.

Vairn, 2018-06-01:
Hi All, 
We are having these issues too, here is a video of what it does for us, 
doesn't matter if full or partial callibration.

https://photos.google.com/share/AF1QipMPhNMV79NpuILcOBJeqC2mfHDlWY_kSHMdT5413X1JGRWyh1YyF8pGUaSWNm8p_Q?key=TWNYcjNjMjdaZlRCcHIydHAtSjFqQW5ZSEJXWUR3

We are on 1.2.7

Vairn, 2018-06-01:
We can get it to work again if we replace all files on the microSD card tho.

ZAMHelmets, 2018-06-01:
This happened again for us as well on the final calibration step - makes calibration extremely time consuming.

AlejandroGarcia92, 2018-06-01:
@Vairn Did you format the microSD card as well?

Vairn, 2018-06-01:
Nope just deleted it all first.
Next time I will make a copy first and see if there are differences 

⁣Sent from BlueMail ​