BCN3D/BCN3DSigma-Firmware #177: Extruder linear advance

pbartko, 2017-12-13:
The current Marlin firmware includes Linear Advance to help improve pressure control within the nozzle during acceleration and deceleration.  Looking at the Configuration_adv.h file, I see a section for Extruder Advance that looks similar but is not activated.  Is there a reason why?  Has anyone experimented with this turned on?

Thanks,

AlejandroGarcia92, 2017-12-14:
We are working on it, but we are getting bad results. It seems to fix those issues but we get so much extruder noise and no successful print.

IdiocyCubed, 2017-12-17:
@AlejandroGarcia92  Are you using the new linear advance from the latest version of Marlin?  I have used the legacy linear advance that is included in the current Sigma firmware and it does not perform well at all.   If I printed very slowly it worked, but like you are saying above it caused a lot of noise on the extruder stepper motor. Have you tried porting the latest stepper routines with linear advance into the Sigma firmware?  I think this will be necessary to add this feature and this should also fix some USB and display comms issues. @WanaGo

AlejandroGarcia92, 2017-12-18:
Yes, we have this kind of issues. We are developing from here: https://github.com/MarlinFirmware/Marlin/tree/1.1.x

This marlin version fix USB comm and we have not adapted marlin for display yet.

Also this adaption will take some time because the name variables are not the same in all cases and some sigma methods must be reworked, etc...

But, now, our priority is to make the linear advance work.

pbartko, 2017-12-18:
Marlin's Linear Advance has problems with Cura.  This has to do with Cura using variable extrusion widths and the Linear Advance calculations expecting a constant line width.  Sometimes there are wild oscillations in the advance calculated as a result.  

In 1.2.7RC, will the use of M900 command set K for the advance calculations or do you have to hard code it in the firmware?  I have both Cura and SC3D and would like to test with both slicers.  SC3D has ability to make the extrusion width constant.

AlejandroGarcia92, 2020-07-21:
Finally, Linear advance has not been added due because it's not work properly with our hardware.