BCN3D/BCN3DSigma-Firmware #178: LCD start to be irresponsive, then it become white, what can be done? Firmware 01-1.1.3

Abl3D, 2017-12-14:
Hi, 
My LCD screen start to be very irresponsive and a few minutes later it become white, and i cant see any information not even the starting "splash screen" BCN3D, it blink twice and then stay white?

I've been thinking to format and reinstall the micro SD LCD files, but i can't found the firmware version 01-1.1.3, and i'am scared to updated to the last one whitout a LCD screen working properly.

Does anyone have an advice for my particular case?
Thanks in advance
Abel

AlejandroGarcia92, 2017-12-14:
Strange behaviour, did you contact with support? Anyway I just attached 1.1.3 which is quite old, I don't recommend it.
[v113.zip](https://github.com/BCN3D/BCN3DSigma-Firmware/files/1558423/v113.zip)

Abl3D, 2017-12-14:
Thanks for the response, unfortunatly, remplacing the firmeware file on the SD card do not fix my issue.

Abl3D, 2018-01-13:
So, my LCD appears to be dead, i finnally choose to switch to the new LDC from R17, wich is far better than the old one. It works great now!

closed 2018-01-15 by AlejandroGarcia92