BCN3D/BCN3DSigma-Firmware #180: Wish for Extrusion Multiplayer in Display

maodch, 2018-01-06:
I really miss the option, to set the extrusion multiplier direct on the display.
Speed, Temperature are given, but not this.

Because, if you're started a print, and suddently the extruder begins to snap back because of over extrusion, you only can set up higher temerature, but if that not helps, you must stop the hole print. Even the over extrusion happend in the middle of the print because of the geometry.

Same for under extrusion.

Please consider this option, because especially for rapid prototyping it would be a big advantage.

Thanks
Ramon

guillemavila, 2018-01-08:
Linked to #64

guillemavila, 2018-01-08:
You can get the trick if printing over USB.

- with Simplify you have an extrusion controller (set to 100% by default)
- you can also get the same effect sending a M221 S[percentage]. Set the flow to 90% with `M221 S90`

Note it will change the flow for both extruders.
Right now do not expect the feature in the display in the short term as it would need new display files.

pgilfernandez, 2018-01-08:
Of course, it does the trick but 99% of us don't have the printer attached to a computer...

maodch, 2018-01-08:
Hi @guillemavila 
And the USB-Connection don't really working well. Because (as I already mentioned in another thread) after connecting and aborting a print, the Printer has some variables still saved and is always on.
After starting a new print, you will get some weird movements, crashes into plates, prints in the air and so on. It's not really done yet to recommend printing with USB.
And also if you're hybernating a pc, the Printer will still be connected and lighted, and even after that, the restarted prints sucks. In any case you have to disconnect the USB and reconnect, to get anoterh print successfully run.

So, I wouldn't recommend the USB way at the current state.