BCN3D/BCN3DSigma-Firmware #181: Stepper E-Steps for Bondtech

maodch, 2018-01-06:
Because I was unsatisfied with the shipped extruders (feeders), I bought the suitable Bondtech extruders for my Sigmas, with a amazing result.

For that I had to adjust the E-Steps from 152 (Shipped value) to 492.45.
Randomly (and without any Firmware update or changes on the machine itself), the value goes lost and goes overwrited with 152 again.

Please eigher tell me, what this "reset" behavior triggers, or even better, please don't automaticly overwrite this value. 

And don't understand me wrong, I love Sigma, all three I have, but the feedder has often to less grip.

Thanks in advance
Ramon

AlejandroGarcia92, 2018-01-08:
This reset can be done by going into Setup Assistant and when the firmware version has changed.

maodch, 2018-01-08:
Hi @AlejandroGarcia92 

Unfortunately in my case, the Firmware Setting changed without any update or Setup assistant use.
Is I wrote, changed the E-Steps (saw with matterhackers) set back to original value, and I printed many hours with another Setup for Bondtech.

I cannot explain why it reset itself.

maodch, 2018-01-12:
I could reproduce the issue. 
If I press "STOP" and "SAVE" the print, the E-Steps are lost after turning on the Sigma again.
So the print will continue, but there is not really filament comming out the nozzle, so I had could canceled the job anyway because it's crap.

Why the Firmware (or the settings) will overwrite itself?

maodch, 2018-01-15:
I stored it with MatterControl - > Configuration -> and saved there to EPROM

AlejandroGarcia92, 2018-01-16:
I've just tested and you are right, I will test to press STOP SAVE.

AlejandroGarcia92, 2018-01-16:
I've test STOP SAVE right now and E steps are not overwritten. Checking the firmware code, the E-steps should be overwritten itself in the setup assistant and when the firmware version is changed.

maodch, 2018-01-16:
That's weird.
You are absolutely right.
Maybe I was to stupid in this case.

Sorry for that.

closed 2018-01-16 by maodch