BCN3D/BCN3DSigma-Firmware #186: URGENT: Stuttering when printing via USB (v1.2.7)

mrcg, 2018-02-18:
I have updated to 1.2.7.
Unfortunately all printer are failing after some while because printer is stuttering (nozzle does not move for some seconds).

In the printing terminal i can see "Display is ready" and "Display got disconnected" all the time.

PLEASE URGENTLY FIX THIS ISSUE !!!!
All this issues when printing via USB are annoying.

Send: N93602 G1 X103.216 Y161.437 E0.0707*93
Recv: Display is ready
Recv: Display got disconnected
Recv: ok
Recv: ok

Recv: Display got disconnected
Recv: ok
Send: N93646 M105*41
Recv: ok

Send: N93706 G1 X111.260 Y156.133 E0.3203*93
Recv: Display is ready
Recv: Display got disconnected
Recv: ok
Send: N93707 G92 E0*77
Recv: ok

tsp666, 2018-02-19:
HI, just an idea, as I am having no problems even with 14 hour jobs through usb - did you ever try to print trough a raspberry with OCTOPRINT on it ? (the only way I am using USB)

Just my 2c..

AlejandroGarcia92, 2018-02-19:
@mrcg There is a timeout between the connection screen-mainboard, but right now i don't know the cause.

mrcg, 2018-02-19:
I am using Octoprint on Raspberry on Sigma since day one (about 2 years now) without such issues.
The issue happens sporadically after some time on dual extruder print.
I flashed back now v1.2.6 to get rid of this problem.
The issues i had and have with USB printing are really frustrating.
This basic feature should be better tested and improved.
Using ocoprint is quite common to add networking capabilities to 3d printer.

thesec, 2018-02-21:
+1 on the request for a fix

Prejkox, 2018-02-25:
+1 on the request for a fix

AlejandroGarcia92, 2018-02-26:
I will investigate how to fix it. I will try to reproduce this issue, I would tell you if I can't.

fjohannes, 2019-11-11:
Have the same issue while printing via Octoprint from a RasPi3.
Sigma R17 with 2.0.8 firmware.
@AlejandroGarcia92 any news on that?

AlejandroGarcia92, 2019-11-12:
We didn't fix it.

fjohannes, 2019-11-12:
Any chance this is a hardware issue I could try to modify (not firm connection, cable shielding, cable size)? Maybe permanently disconnect the LCD /add a hardware switch to turn it off?
Just for clarification: is the message related to the hardware/machine LCD?
Or rather a software problem?

nachtbraker, 2019-11-12:
Just to add a datapoint: I’m running Octopi just fine on a RPi3b and a Sigma R17.

Maybe this is can help: I have seen stuttering in the past when I first started using OctoPi. It turned out that I had the Sigma’s LCD screen set to the temperature display. That interfered with proper serial communication I guess. Nowadays I always make sure the display shows the big USB symbol and I have no problem with stuttering.

Ben

fjohannes, 2019-11-12:
The display on my Sigma shows the USB-Symbol ("Serial Print") by default as long as Octoprint is running a print. Maybe because of my later firmware? Which firmware are you running @nachtbraker ?

nachtbraker, 2019-11-12:
Currently 01-2.0.7, but the stuttering I had in the past was obviously with an older version. At that point, I do not think the USB symbol was visible perhaps. The only reason I’m bringing it up is that this was a repeatable scenario in which I could trigger the stuttering. Hopefully this information helps a bit for those concerned.

Ben