BCN3D/BCN3DSigma-Firmware #188: [enhancement] Support for full name readability in the file browser

Prejkox, 2018-02-25:
I have had a problem with that it is only possible to read 22 letters of the full name of the gcode files on the sigma since I bought it. A printer in this price segment really should be able to show the whole file name even if it is longer then 22 letters. It is really hard to to choose the correct file if I have done the same thing with small changes and added that in the end of the file name. 

I propose that when you tap the file name it should begin to scroll after a few seconds and keep on doing that. Then when you know that it is the correct file you tap on the file name once more and then I get to the print screen. 
In fact it does not matter to me how it is done the way to show the whole file name as long as it is possible. The above idea is just a proposal.

BR T

AlejandroGarcia92, 2018-02-26:
I know it, but it is for space.

Prejkox, 2018-03-08:
I did not understand what you wrote in the last post AlejandroGarcia92. Can you explain it further?

AlejandroGarcia92, 2018-03-09:
Sorry,

What I wanted to say is that we shouldn't put more letters because the typo size should be smaller.

I didn't give you an answer for your proposals, sorry. 

According to your proposals, we have display limitations for the implementation of the features that you proposed.

closed 2018-09-24 by AlejandroGarcia92

Prejkox, 2018-10-08:
@AlejandroGarcia92 I see that you closed this query. I have read your latest answer but still do not fully understand and was wondering if you misunderstood my wish. 
I understand that the screen can not present more letters. But there must be some way to be able to present a longer filename then the printer currently do. You are talking of display limitations. Is it not possible to scroll a file name on the screen? When I write scroll I mean that the file name start to move from to the left in screen erasing the first letters and exposing the letters in the end that earlier was not possible to see. This is how it works on Ultimaker3 file browser, they also cant display the full file name and have solved it like this. Anyhow I just got the feeling that you where just focusing on that the number of letter that where presented and I get that it is not possible to squeeze in more letters then it is right now.
If it is not possible to scroll/move the file name as I tried explain when marking the filename, could it be presented with more letters in the next screen that appears when clicking the filename? where you shall click cancel or ok?
OR could it be possible to add a screen directly after clicking on the filename with only the file name presented that you just need to touch anywhere and it moves on to the "do you want to print this file screen or not" that is in the printer right now. IF that is possible you can use the whole screen to present the file name on multiple rows.

I'm sure that there is ways to do this without going out of the bounds of what the screen can handle. It is just so annoying when I have several iterations of a print and there is no way to see which one that I should choose. This is a professional printer and it is a bit strange that it is not possible to be able read the full filename in any way. At work we are committed to name the files in a certain manner with a number serie and the iterations in the end of the article number but due to this issue we cannot separate the prints. We just can sort out the latest file out due to it is on the last row in the browser. To many times I began to print the wrong file if I want to reprint an earlier version due to this issue.

Please look in to this limitation further.

BR Prejkox