BCN3D/BCN3DSigma-Firmware #189: Homing Z should home X also

Loimulohi, 2018-02-26:
I issued Homing Z command from Simplify3D and both printheads were above print bed and not in parking position. Left one went to parking position and then went to position to use that microswitch. Right printhead stayed where it was and print bed rammed to right nozzle. I had to switch off the printer and cancel Z homing command.

So, should that Homing Z command to be modified that both printheads would move to parking position before moving print bed up, just in case someone has forgotten printheads above the print bed?

AlejandroGarcia92, 2018-02-26:
It could be implemented, even a all home procedures could be reworked.

AlejandroGarcia92, 2020-07-21:
#226

closed 2020-07-21 by AlejandroGarcia92