BCN3D/BCN3DSigma-Firmware #190: change the filament during printing BUG

hst68, 2018-03-09:
If you change the filament during printing (Pause ->Tools->Change Filament. If it happens again and again that the process ends when unloading, but the LCD screen gets stuck with the process view. Actually, he should go to Lade Filament Screen. That's annoying because every pressure is destroyed! Firmware 1.2.7 R17

AlejandroGarcia92, 2018-03-09:
We had this issue with bondtech and we saw that the relation between steps/mm and max feedrate was too high for the microcontroller and we set the max feedrate of E motors to 40mm/s for Sigmax which works with bondtech(Sigma has set by default 60). 

Use the following commands:
M203 E40.00
M500

hst68, 2018-03-09:
Great, it works!

closed 2018-03-12 by AlejandroGarcia92