BCN3D/BCN3DSigma-Firmware #193: Sudden shift of the x-Axis

svengl, 2018-06-18:
Another print, another failure

During print with the right extruder, the x-axis suddenly shifts. This threw off the whole build plate and  resulted in a destroyed z-Axis end switch...! Some black plastic parts broke off.
This time Simplify3D was used due to bug https://github.com/BCN3D/Cura/issues/105 but I am pretty sure it is not a slicer issue.

[image]
[image]
[image]
[image]
[image]
[image]

AlejandroGarcia92, 2018-06-18:
It seems that X right motor stepper driver is losing steps.

svengl, 2018-06-18:
So what can I do? We started another print with a new end switch and it works fine at the moment. 
Could a replacement of the stepper driver boards help? Or is that improbable? Or rather the mainboard?

sennalb, 2018-06-29:
Hi,
I have the same problem with sigmax and the top trasparent cover, but if i remove the top cover the problem does not arise again.

I suppose that is temperature problem, I using the printer in my office with an ambient temperature of about 20-25 degrees. With the top cover the internal temperature of sigmax printer it seems too high: the bed is hot (65 deggres) the temperature of the motors of X axis is very very high.

My 2 cents
Alberto Senni
Studio Tecnico Zocca BO (Italy)

AlejandroGarcia92, 2018-07-02:
Did you contact with support? They are dealing with those kind of issues.

sennalb, 2018-07-06:
Yes, I am in contact with support
Thanks 
Alberto

closed 2018-07-20 by AlejandroGarcia92

svengl, 2018-08-17:
Maybe a closing comment on this:
As somehow expected, it was not the SD Card reader (as suggested in https://github.com/BCN3D/BCN3DSigma-Firmware/issues/155) 
We sent the printer to our dealer and they found that the stepper driver voltage was set way too high, so they adjusted it. Further, the right stepper driver was exchanged as it was probably broken.
Now it seems to be running well for the time being.