BCN3D/BCN3DSigma-Firmware #195: Can't Compile

crazyp1g, 2018-08-05:
Arduino:1.8.5 (Windows Store 1.8.10.0) (Windows 10), 开发板："Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\Hysteresis.cpp:27:0:

sketch\language.h:109:25: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

 #define MSG_M115_REPORT "FIRMWARE_NAME:Marlin "MACHINE_NAME"; FIRMWARE_VERSION:"VERSION_STRING"; FIRMWARE_URL:" FIRMWARE_URL "; PROTOCOL_VERSION:" PROTOCOL_VERSION "; MACHINE_TYPE:" MACHINE_NAME "; EXTRUDER_COUNT:" STRINGIFY(EXTRUDERS) "; UUID:"

                         ^

sketch\language.h:109:60: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

 #define MSG_M115_REPORT "FIRMWARE_NAME:Marlin "MACHINE_NAME"; FIRMWARE_VERSION:"VERSION_STRING"; FIRMWARE_URL:" FIRMWARE_URL "; PROTOCOL_VERSION:" PROTOCOL_VERSION "; MACHINE_TYPE:" MACHINE_NAME "; EXTRUDER_COUNT:" STRINGIFY(EXTRUDERS) "; UUID:"

                                                            ^

In file included from sketch\LCD_FSM.h:23:0,

                 from sketch\LCD_FSM.cpp:6:

sketch\language.h:109:25: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

 #define MSG_M115_REPORT "FIRMWARE_NAME:Marlin "MACHINE_NAME"; FIRMWARE_VERSION:"VERSION_STRING"; FIRMWARE_URL:" FIRMWARE_URL "; PROTOCOL_VERSION:" PROTOCOL_VERSION "; MACHINE_TYPE:" MACHINE_NAME "; EXTRUDER_COUNT:" STRINGIFY(EXTRUDERS) "; UUID:"

                         ^

sketch\language.h:109:60: warning: invalid suffix on literal; C++11 requires a space between literal and string macro [-Wliteral-suffix]

 #define MSG_M115_REPORT "FIRMWARE_NAME:Marlin "MACHINE_NAME"; FIRMWARE_VERSION:"VERSION_STRING"; FIRMWARE_URL:" FIRMWARE_URL "; PROTOCOL_VERSION:" PROTOCOL_VERSION "; MACHINE_TYPE:" MACHINE_NAME "; EXTRUDER_COUNT:" STRINGIFY(EXTRUDERS) "; UUID:"

                                                            ^

sketch\LCD_FSM.cpp: In function 'void lcd_fsm_lcd_input_logic()':

sketch\LCD_FSM.cpp:445:70: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

    genie.WriteStr(STRING_UTILITIES_FILAMENT_CHANGEFILAMENT_TEMPS,"0%");

                                                                      ^

sketch\LCD_FSM.cpp:701:70: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

    genie.WriteStr(STRING_UTILITIES_FILAMENT_CHANGEFILAMENT_TEMPS,"0%");

                                                                      ^

sketch\LCD_FSM.cpp:1636:71: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

     genie.WriteStr(STRING_UTILITIES_FILAMENT_CHANGEFILAMENT_TEMPS,"0%");

                                                                       ^

sketch\LCD_FSM.cpp:1999:70: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

    genie.WriteStr(STRING_UTILITIES_FILAMENT_CHANGEFILAMENT_TEMPS,"0%");

                                                                      ^

sketch\LCD_FSM.cpp:2115:72: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

      genie.WriteStr(STRING_UTILITIES_FILAMENT_CHANGEFILAMENT_TEMPS,"0%");

                                                                        ^

sketch\LCD_FSM.cpp:2403:70: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]

    genie.WriteStr(STRING_UTILITIES_FILAMENT_CHANGEFILAMENT_TEMPS,"0%");

                                                                      ^

[... 552 lines cut ...]
                 from sketch\genieArduino.h:49,

                 from sketch\Marlin.h:30,

                 from sketch\stepper.cpp:24:

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino/HardwareSerial.h:101:30: error: 'volatile uint8_t* const HardwareSerial::_udr' is protected

     volatile uint8_t * const _udr;

                              ^

stepper.cpp:221: error: within this context

    *MYSERIAL._udr;

              ^

exit status 1
within this context

在文件 -> 首选项开启
“编译过程中显示详细输出”选项
这份报告会包含更多信息。
 what's the matter？

crazyp1g, 2018-08-05:
i just download and no any modify

dvb-adenin, 2018-08-11:
that's why:
```
sketch\stepper.cpp:182:18: note: #pragma message: Hardware Serial must not have any protected variables or this will not work! Modify HardwareSerial.h to be public!
```
solution:
edit **C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.10.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino/HardwareSerial.h**
search this :
```
class HardwareSerial : public Stream
{
  protected:
```
change **protected:** to **public:**

looks somewhat brutal, but it works

closed 2018-09-25 by AlejandroGarcia92